RobotLocomotion/drake #3449: quat2axis does not work properly for the degenerate case

hongkai-dai, 2016-09-14:
currently `quat2axis` in `drake/math/quaternion.h` does not work for the degenerate case, when the rotation angle is 0 degree. Specifically it returns the rotation axis to be a zero vector.
```
cout<<quat2axis(Vector4d(1.0, 0.0, 0.0, 0.0))
```
the output is
```
0
0
0
0
```
On the other hand, in the other code, we assumes that the axis is normalized, or at least is not a zero vector. For example, in `axis2rotmat`, we did
```
const auto& axis = (a.template head<3>()) / (a.template head<3>()).norm();
```

I would propose to treat the 0 degree as a special case, such that the axis returned from `quat2axis` always has a unit length, for example, `[1;0;0]`. As suggested by http://www.euclideanspace.com/maths/geometry/rotations/conversions/quaternionToAngle/

This is the singularity case of axis-angle representation, that the function`quat2axis` will not be differentiable.

sherm1, 2016-09-14:
:+1: -- quat2axis() should promise always to return a unit vector and angle. Returning `[1 0 0], 0` in the singular case is as good as any other axis. One tricky issue is deciding at what tolerance to give up on calculating an axis; let's discuss.

sherm1, 2016-09-14:
Here is the Simbody implementation -- note that `acos()` should not be used!

```c++
//------------------------------------------------------------------------------
// Returns [ a vx vy vz ] with (a,v) in canonical form, i.e., -180 < a <= 180 and |v|=1.
// The cost of this operation is roughly one atan2, one sqrt, and one divide (about 100 flops).
//------------------------------------------------------------------------------
template <class P> Vec<4,P>
Quaternion_<P>::convertQuaternionToAngleAxis() const {
    const RealP& ca2  = (*this)[0];       // cos(a/2)
    const Vec3P& sa2v = this->template getSubVec<3>(1);  // sin(a/2) * v
    RealP        sa2  = sa2v.norm();      // sa2 is always >= 0

    const RealP Eps = NTraits<P>::getEps();
    const RealP Pi  = NTraits<P>::getPi();

    // TODO: what is the right value to use here?? Norms can be
    // much less than eps and still OK -- this is 1e-32 in double.
    if( sa2 < square(Eps) )  return Vec4P(0,1,0,0); // no rotation, x axis

    // Use atan2.  Do NOT just use acos(q[0]) to calculate the rotation angle!!!
    // Otherwise results are numerical garbage anywhere near zero (or less near).
    RealP angle = 2 * std::atan2(sa2,ca2);

    // Since sa2>=0, atan2 returns a value between 0 and pi, which is then
    // multiplied by 2 which means the angle is between 0 and 2pi.
    // We want an angle in the range:  -pi < angle <= pi range.
    // E.g., instead of rotating 359 degrees clockwise, rotate -1 degree counterclockwise.
    if( angle > Pi ) angle -= 2*Pi;

    // Normalize the axis part of the return value
    const Vec3P axis = sa2v / sa2;

    // Return (angle/axis)
    return Vec4P( angle, axis[0], axis[1], axis[2] );
}
```

tkoolen, 2016-09-15:
> ```
const auto& axis = (a.template head<3>()) / (a.template head<3>()).norm();
```

Woah, I think that's actually a bug! We're storing a reference to a temporary here!

david-german-tri, 2016-09-15:
> Woah, I think that's actually a bug! We're storing a reference to a temporary here!

I don't think it's a bug, due to const reference lifetime extension, but for sure it's unnecessarily confusing.

https://herbsutter.com/2008/01/01/gotw-88-a-candidate-for-the-most-important-const/

tkoolen, 2016-09-15:
Aha, thanks, learned something.

PR #3604 (2016-10-07): Add rotation conversion test
This PR does two things
1. Partially resolves #3427. I will push more PRS to clean up the remaining parts of `drakeGeometryUtil`.
2. Resolves #3449 #3451 . The test `rotation_conversion_test` tests the singular and non-singular cases.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3604)
closed 2016-10-07 by hongkai-dai