RobotLocomotion/drake #3451: rotmat2axis does not work for 180 degrees of rotation

hongkai-dai, 2016-09-14:
There are two degenerate cases for axis-angle representation, at 0 degree and 180 degrees. `rotmat2axis` in `drake/math/rotation_matrix.h`handles 0 degree as a special case, but does not handle 180 degree. So if I use the rotation matrix corresponding to 180 degree rotation about z axis, the return value from `rotmat2axis` is a rotation of 180 degree about **zero** vector. 

We need to implement the corner cases handled in 
http://www.euclideanspace.com/maths/geometry/rotations/conversions/matrixToAngle/

sherm1, 2016-09-15:
Per f2f with @hongkai-dai, a very robust and efficient way to do this would be: fix `quat2axis()` as discussed in #3449, then use Drake's existing rotation matrix to quaternion method followed by `quat2axis()`.

hongkai-dai, 2016-09-15:
FWIW, I also tested the implementation in http://www.euclideanspace.com/maths/geometry/rotations/conversions/matrixToAngle/, the numeric performance is not as good as `quat2axis(rotmat2quat(R))`, since it calls `acos` in that implementation. And as @sherm1 points out, `acos` is not numerically stable near 1.

hongkai-dai, 2016-09-15:
@rdeits , this could be the reason why Atlas falls over in the DRC on the second day, that we did not handle this 180 degree rotation properly.

closed 2016-10-12 by hongkai-dai