RobotLocomotion/drake #3466: `message` may be used uninitialized

mwoehlke-kitware, 2016-09-15:
I am getting several "errors" (due to use of some additional `-Werror` local to my build) similar to:

```
In file included from /usr/local/build/matthew/work-mit/drake/install/include/lcmtypes/drake/lcmt_simple_car_state_t.hpp:7:0,
                 from /usr/local/src/matthew/work-mit/drake/drake/../drake/automotive/gen/simple_car_state.h:13,
                 from /usr/local/src/matthew/work-mit/drake/drake/../drake/automotive/gen/simple_car_state_translator.h:6,
                 from /usr/local/src/matthew/work-mit/drake/drake/automotive/gen/simple_car_state_translator.cc:1:
/usr/local/build/matthew/work-mit/drake/install/include/lcm/lcm_coretypes.h:278:29: error: ‘message’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
         buf[pos++] = (v>>56)&0xff;
                             ^
/usr/local/src/matthew/work-mit/drake/drake/automotive/gen/simple_car_state_translator.cc:24:34: note: ‘message’ was declared here
   drake::lcmt_simple_car_state_t message;
                                  ^
cc1plus: some warnings being treated as errors
```

In fact, the "may" here is wrong; the class *is* being used (partially) uninitialized. Specifically, a new message is being created in these functions (i.e. the same issue exists in all of the translators), but the `timestamp` field is not being set.

It would be appreciated if someone familiar with this code (@jwnimmer-tri?) can determine how `timestamp` is supposed to be initialized here and fix it. Also, I wonder if we should add `-Werror=maybe-uninitialized` to everyone's build, since it clearly caught a real error here...

mwoehlke-kitware, 2016-09-15:
p.s.

```
In file included from /usr/local/src/matthew/work-mit/drake/drake/automotive/test/simple_car_state_translator_test.cc:6:0:
/usr/local/build/matthew/work-mit/drake/install/include/gtest/gtest.h: In instantiation of ‘testing::AssertionResult testing::internal::CmpHelperGT(const char*, const char*, const T1&, const T2&) [with T1 = long unsigned int; T2 = int]’:
/usr/local/src/matthew/work-mit/drake/drake/automotive/test/simple_car_state_translator_test.cc:28:173:   required from here
/usr/local/build/matthew/work-mit/drake/install/include/gtest/gtest.h:1530:145: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
 GTEST_IMPL_CMP_HELPER_(GT, >);
```

...needs:

```diff
diff --git a/drake/automotive/test/simple_car_state_translator_test.cc b/drake/automotive/test/simple_car_state_translator_test.cc
index 2f83a18..13ff0b8 100644
--- a/drake/automotive/test/simple_car_state_translator_test.cc
+++ b/drake/automotive/test/simple_car_state_translator_test.cc
@@ -25,7 +25,7 @@ GTEST_TEST(SimpleCarStateTranslatorTest, RoundtripTest) {
   // Encode the message.
   std::vector<uint8_t> lcm_message_bytes;
   dut.TranslateVectorBaseToLcm(publish_state_vector, &lcm_message_bytes);
-  EXPECT_GT(lcm_message_bytes.size(), 0);
+  EXPECT_GT(lcm_message_bytes.size(), 0u);
 
   // Decode the message.
   SimpleCarState<double> subscribe_state_vector;
```

(This *should* be showing up on CI; not sure why it isn't...)

jwnimmer-tri, 2016-09-15:
I believe setting the timestamp is fixed via #3460.

jwnimmer-tri, 2016-09-15:
P.S., I am all for more error flags being turned on.  Someone just has to shave the yaks to make it happen.

mwoehlke-kitware, 2016-09-15:
Okay, tomorrow I may look into making a PR to turn on `-Wmaybe-uninitialized`; this is hopefully the only spot tripping it. (It seems to be the only spot on my machine, but I don't have many externals turned on...)

mwoehlke-kitware, 2016-09-19:
Well, this is odd... `-Wmaybe-uninitialized` should be included in `-Wall`... seems for some reason only my computer is throwing the warning. I still believe it to be genuine, but looks like CI won't catch it :cry:. I'll check if it's working now that #3460 is merged...

mwoehlke-kitware, 2016-09-30:
Fixed by #3460

closed 2016-09-30 by mwoehlke-kitware