RobotLocomotion/drake #3467: Viewing control signals from simulation

samkhal, 2016-09-15:
I'm running the Atlas demos and want to be able to analyze the control torques that are being sent. I looked at https://github.com/RobotLocomotion/drake/issues/1847, but that didn't fully answer my question.

In that issue, Russ mentions that the signals are output directly by `simulate`; however, the xtraj and ytraj outputs of simulate appear to be identical (`isequal` returns 1); they both contain the position+velocity state at each timestep, but not the control. Is there any way to capture the control signals directly from simulink without LCM?

With LCM logging, is there a way to avoid the crash mentioned in the issue I linked? I'm in the process of installing R2014a as per the suggestion, but I'd rather not need to downgrade if possible.

RussTedrake, 2016-09-16:
you can always add the torques to an output from the wiring diagram level by modifying the mimo feedback and mimo cascase jumble.  (this will be much cleaner in the new c++ systems)

samkhal, 2016-09-16:
Thanks, that's what I've been doing; I thought there might be a built-in way to do this more cleanly.

As a side note, any ballpark estimate/roadmap on when the shift from matlab to c++ would be complete? I'm looking forward to it.

RussTedrake, 2016-09-16:
The c++ functionality is improving rapidly.  We expect to have the atlas / valkyrie walking pipeline moved to c++ completely probably in ballpark one month

closed 2016-09-16 by RussTedrake