RobotLocomotion/drake #3470: Eigen quaternion elements do not lay out in memory in the same way Drake does.

amcastro-tri, 2016-09-16:
We make an extensive use of Eigen in Drake. It makes sense that `Quaternion`'s are also used instead of just 4d vectors. Eigen already provides convenience conversion between Quaternions, AxisAngle, 3d matrices and Isommetries and we could leverage on them.

However Eigen layout in memory of entries is different from Drake's. 
Eigen layout is: first three components are the vector part and the last component corresponds to the scalar part of the Quaternion.

In Drake this is the opposite: the first entry corresponds to the scalar part of the quaternion while the last three components correspond to the vector component.

When trying to use Eigen quaternions from Drake's internals Vector4's this forces us to create unnecessary copies to deal with the layout.

jwnimmer-tri, 2016-09-16:
Relevant: [drake/math/quaternion.h#L189](https://github.com/RobotLocomotion/drake/blob/0be1d5bfd3a17797a4463fbbb0d24076c2af9e97/drake/math/quaternion.h#L189).

sherm1, 2016-09-16:
We already use Eigen's rotation matrix class (Isometry3) -- we should definitely be using Eigen's Quaternion class rather than a roll-you-own version!

jwnimmer-tri, 2016-09-16:
If I'm reading everything correctly, I think the only real problem is this code from #3471:
```
void VectorOfPoses<T>::set_body_orientation(int body_index,
                                             const Quaternion<T>& q) {
  const int body_start = 7 * body_index;
  this->get_mutable_value().template segment<4>(body_start) = q.coeffs();
```

This has `VectorOfPoses` storing poses in ((x y z w) (x y z)) order, which is wrong.  It should be ((w x y z) (x y z)) order, in which case everything is consistent and the issue can be closed.

sherm1, 2016-09-16:
Another reason to use Eigen's Quaternion class is issue #3449 where Drake's homebrew quaternion-to-angleaxis uses a numerically unstable method (apparently grabbed from the internet somewhere) while Eigen's implementation [here](https://bitbucket.org/eigen/eigen/src/e4ccd78402d5d21317540f2d5d17d878de1484e4/Eigen/src/Geometry/AngleAxis.h?at=default&fileviewer=file-view-default#AngleAxis.h-167) is correct.

hongkai-dai, 2016-09-16:
I would say we should migrate to Eigen's Quaternion, AngleAxis and Isometry3d matrix. We should actually deprecate the method inside `math/rotation_matrix.h, math/axis_angle.h`and `math/quaternion.h`. The tricky part is the roll, pitch yaw representation. Drake uses z-y'-x'' rotation axes, but Eigen does not have this representation. Instead, Eigen uses the Euler angles, defined as z-x'-z'' rotation, http://eigen.tuxfamily.org/dox/group__Geometry__Module.html#ga0db1cc067c51aaa6dedf5805ee0c53d7

sherm1, 2016-09-16:
Looks like Eigen has a general rotation matrix-to-arbitrary Euler sequence method [here](http://stackoverflow.com/questions/27508242/roll-pitch-and-yaw-from-rotation-matrix-with-eigen-library). Where does it specify the z-x'-z'' sequence? (BTW the term "Euler angles" can also mean any generic 3-angle sequence.)

hongkai-dai, 2016-09-16:
@sherm1 , you are right, I misread the example in Eigen documentation [here](http://eigen.tuxfamily.org/dox/group__Geometry__Module.html#ga0db1cc067c51aaa6dedf5805ee0c53d7). The documentation does not say if it is using intrinsic or extrinsic order. But I assume we can test if
```
Vector3f ea = mat.eulerAngles(0, 1, 2); 
```
is the inversion of
```
mat = AxisAnglef(ea(0), Vector3f::UnitX())
         *AxisAnglef(ea(1), Vector3f::UnitY())
         *AxisAnglef(ea(2), Vector3f::UnitZ())
```

RussTedrake, 2016-09-16:
agreed that we should conform to eigen. 
one pain point is that the matlab code uses the old representation (as @hongkai-dai points out), so we will have to be careful at every entry/exit point, or update the matlab.

hongkai-dai, 2016-09-16:
I have spent two days trying to clean up `drakeGeometryUtil`, which contains many conversion between different rotation representations. I found several bugs in our code, as mentioned in #3449 , #3451 . Basically we did not test these conversions for the corner cases

1. `rotmat2axis` does not work for 180 degrees rotation.
2. `quat2axis` does not work for 0 degree rotation.
3. `rotmat2rpy` does not work for pitch = pi/2

There can be more corner cases that I have not digged out. 

In this first round of cleanup, I am just rewriting the conversions in `drakeGeometryUtil`, to make sure the code works for these corner cases. I will send the PR soon.

I think after this PR, we should just migrate to `Eigen/geometry` module. In MATLAB, we should create class `Quaternion`, `RotationMatrix`, `AxisAngle`, which are just a thin layer that wraps the C++ classes. They should just contain a mex pointer.

tkoolen, 2016-09-19:
I'm in favor of moving to `Eigen::Quaternion`. I just wanted to add that I think the conversion should be done in one fell swoop (as a separate pull request), as opposed to the current strategy of  starting to use `Eigen::Quaternion` in new pull requests while old code still uses `Eigen::Vector4d`. This latter strategy will lead to lots of confusion and room for bugs.

hongkai-dai, 2016-09-19:
If we are going to completely change to `Eigen::Quaternion`, then it also makes sense to use `Eigen::AngleAxis` and `Eigen::EulerAngles` (the latter is in `unsupported/Eigen/EulerAngles`). 

In the PR #3502, the API still uses `Eigen::Vector4d` to represent a quaternion. It only calls `Eigen::Quaternion` inside the conversion function. For example, when converting quaternion to roll pitch yaw, the function is
```
Vector3d quat2rpy(const Eigen::Vector4d &quat) {
// transform drake quaternion to Eigen quaternion.
Eigen::Quaternion<typename Derived::Scalar> q_eigen(q(0), q(1), q(2), q(3));
// compute Eigen euler angles
Eigen::Matrix<typename Derived::Scalar, 3, 3> rotmat =q_eigen.toRotationMatrix();
auto euler_angles =Eigen::EulerAngles<typename Derived::Scalar, Eigen::EulerSystemZYX>::template FromRotation<false, false, false>(rotmat);
// convert Eigen euler angles to drake roll pitch yaw convention
return drake::Vector3<typename Derived::Scalar>(euler_angles.gamma(), euler_angles.beta(), euler_angles.alpha());
}
```
I believe by restricting the usage of Eigen quaternion inside the conversion function, the confusion is minimized.

The PR #3502 is meant to clean up the drakeGeometryUtil file with minimal change. After this PR, I will submit a new PR, that changes to `Eigen::Quaternion`, `Eigen::AngleAxis` and `Eigen::EulerAngles`.

amcastro-tri, 2017-03-08:
What do you guys think if we instead start using `unsupported/Eigen/EulerAngles`? We could introduce them into Drake with our own unit tests to verify correctness and support for Autodiff. 
`AutoDiff`  is also an unsupported module and we still use it.

hongkai-dai, 2017-03-08:
I talked with them about our unit test, and our desire to use Autodiff. They prefer the randomized test, and they do not think they should support autodiff, since autodiff is in the unsupported module also.

amcastro-tri, 2017-03-08:
So that code doesn't even compile with AutoDiffScalar?

hongkai-dai, 2017-03-08:
We do not know yet, the `Eigen/EulerAngles` is never tested against AutoDiffScalar, it is only tested with double and float.

jwnimmer-tri, 2018-12-21:
I think it's fair to say that @mitiguy should have a read here, and tell us what to do.  My guess is that this issue should be closed.

richardrl, 2023-01-20:
It would be a good idea to include the quaternion format in the Drake documentation, whether the real part comes first or last

sherm1, 2023-01-20:
@richardrl where in the documentation would that have been helpful for you?