RobotLocomotion/drake #3476: move drakeURDF.html to drake.mit.edu

RussTedrake, 2016-09-16:
Related to #2881 .

It looks like drake002 is still used to host the urdf xml schema documentation (which may be out of date by now).  is that already being generated by the AWS host?  if not, what are the remaining steps?

liangfok, 2016-09-16:
I assume this issue is referring to http://drake002.csail.mit.edu/drake/urdf/drakeURDF.html, which is linked to from http://drake.mit.edu/?

RussTedrake, 2016-09-16:
correct.

jamiesnape, 2016-09-16:
> is that already being generated by the AWS host?

No, it is not. Do you have a script to generate it?

RussTedrake, 2016-10-10:
it was in the makefile that got nuked in the doc directory.  
```
	ifeq ($(shell uname -p),Darwin)
	  cd doc && /Applications/oxygen/schemaDocumentationMac.sh drakeURDF.xsd -cfg:oxygen_export_settings_html.xml	
	else
	  # note that I manually installed oxygen to my home oxygen directory
	  cd doc && $(HOME)/oxygen/schemaDocumentation.sh drakeURDF.xsd -cfg:oxygen_export_settings_html.xml	
	endif 
```

david-german-tri, 2017-01-13:
Done in https://github.com/RobotLocomotion/drake-ci/pull/72

closed 2017-01-13 by david-german-tri