RobotLocomotion/drake #3483: Consider Simplifying LCM Translator Methods

liangfok, 2016-09-17:
Currently, LCM Translators have the following two methods:

1. `LcmAndVectorBaseTranslator::TranslateLcmToVectorBase()`
2. `LcmAndVectorBaseTranslator::TranslateVectorBaseToLcm()`

I'm thinking about simplifying it to be:

1. `LcmAndVectorBaseTranslator::TranslateFromLcm()`
2. `LcmAndVectorBaseTranslator::TranslateToLcm()`

Thoughts?

liangfok, 2016-09-17:
Actually, another idea is to generalize the translators to support more than just LCM. I'm going to update this issue to reflect this new idea.

liangfok, 2016-09-17:
Done. @jwnimmer-tri, since you consulted on the original design, I am specifically seeking your opinion.

jwnimmer-tri, 2016-09-17:
IMO, both new name proposals are superior to the existing names.
I think it's a toss-up for which new pair is better, so you get to choose.

Nit: `DeSerialize` should be just `Deserialize`.

liangfok, 2016-09-17:
OK thanks. I'll probably go with `Serialize` / `Deserialize` since that's middware agnostic. The middleware will be implied by the name of the translator class.

PR #3497 (2016-09-19): Simplified LCM Translator Method Names
Renamed "TranslateVectorBaseToLcm" to be "Serialize" and "TranslateLcmToVectorBase" to be "Deserialize".

Resolves #3483.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3497)
closed 2016-09-19 by jwnimmer-tri