RobotLocomotion/drake #3492: RigidBodyPlant Unit Test Should Not Rely on Parsers

liangfok, 2016-09-18:
# Problem Definition

The unit test for `RigidBodyPlant`is:

[drake-distro/drake/systems/plants/rigid_body_plant/test/rigid_body_plant_test.cc](https://github.com/RobotLocomotion/drake/blob/131f6f42d0041c910bacee5e7d186c498d137405/drake/systems/plants/rigid_body_plant/test/rigid_body_plant_test.cc)

It currently uses Drake's URDF parser to load the `RigidBodyTree` within the `RigidBodyPlant`. This makes the unit test brittle since it will fail on parser-specific bugs.

A second problem is we want to eventually decouple the parsers from the multi-body dynamics model (which includes the `RigidBodyTree`). Our goal is to allow the software libraries that contain the parsers to link against the libraries that contain the multi-body dynamics model, sensors, actuators, etc., but not the other way around.

# Action Item

Remove the use of the parser from the unit test. Programmatically initialize the `RigidBodyTree` within the `RigidBodyPlant` using its C++ API.

RussTedrake, 2016-09-18:
I don’t see that as a problem?  perhaps we need coverage for the c++ api, but what you describe sounds to me like the test gives coverage to both the tree and the parser.

liangfok, 2016-09-18:
My motivation for creating this issue is based on @jwnimmer-tri's comment here: https://github.com/RobotLocomotion/drake/pull/3228#issuecomment-243647053.

Specifically, the statement:

> The unit tests should not be loading sdfs (the parser should be out of scope); the RBTs should be constructed programmatically.

nkoenig, 2016-09-22:
I try to resolve this pull request.

jwnimmer-tri, 2016-12-22:
Wontfix.

closed 2016-12-22 by jwnimmer-tri