RobotLocomotion/drake #3493: Extract URDF and SDF Parsers from RigidBodySystem

liangfok, 2016-09-18:
# Problem Definition

[RigidBodySystem](https://github.com/RobotLocomotion/drake/blob/5fb0e2537e6bb074ecc0ecc2717364b847ca7782/drake/systems/plants/RigidBodySystem.h) contains URDF and SDF parsers that load modeling elements like sensors and force elements that are external of the [RigidBodyTree](https://github.com/RobotLocomotion/drake/blob/2bce50fd01ced9497a2511f917cc6b7ceb18aa09/drake/systems/plants/RigidBodyTree.h). This is a problem because it will soon be deprecated in favor of System 2.0 systems like [RigidBodyPlant](https://github.com/RobotLocomotion/drake/blob/8b0484208a99a081885f5be14aeef6a2cc35c3ab/drake/systems/plants/rigid_body_plant/rigid_body_plant.h).

# Action Item

Extract the URDF and SDF parsers from `RigidBodySystem`. Move the code into the existing URDF and SDF parsers, which are in:

* [drake-distro/drake/systems/plants/parser_urdf.h](https://github.com/RobotLocomotion/drake/blob/ad24abbd7213876169e978fbb383d0561f99e199/drake/systems/plants/parser_urdf.h)
* [drake-distro/drake/systems/plants/parser_sdf.h](https://github.com/RobotLocomotion/drake/blob/ad24abbd7213876169e978fbb383d0561f99e199/drake/systems/plants/parser_sdf.h)

This will likely involve initially augmenting the parser APIs to take a `RigidBodySystem` as an output parameter. Longer term, the parsers and System 2.0 systems should be decoupled by some intermediate representation that's produced by the parsers and consumed by the System 2.0 systems.

liangfok, 2016-10-17:
I will work more on this after #3824 is merged.

amcastro-tri, 2016-10-26:
I would add to this that probably a good idea would be to have separate files/classes/methods for parsing `RigidBodyTree`'s and for parsing `RigidBodyPlant`'s.

liangfok, 2016-10-26:
From a F2F meeting with @amcastro-tri, I understand the desire is an option where the parser only returns _one_ of the simulation objects (i.e., `RigidBodyPlant`, `RigidBodySensor`, and `RigidBodyForceElement`, etc.) as opposed to _all_ of the objects. This should be trivial to achieve through parameter flags.

sherm1, 2016-10-26:
Better IMO to parse once from file into memory, then provide accessors that RBPlant etc. can use to construct themselves. The parsers should not have to know how we split up the various objects among our classes since that can and will change.

liangfok, 2016-10-26:
Agreed. This gets back to the "intermediate representation" discussion where the intermediate representation decouples the information contained within the modeling files and the simulation objects that are instantiated based on this information. As a first cut, we'll probably end up with the intermediate representation _being_ the simulation objects. However, over time, we may end up deciding to actually decouple the two.

sherm1, 2016-10-26:
A plausible parser target would be the in-memory tinyXml2 form of SDF! That may already have all the ingredients we need. If we can parse both urdf and sdf into that form, we could use it as an in-memory intermediate right now. Maybe someday we would want something fancier but it wouldn't have to be more than that to decouple parsing from construction. We could wrap a thin parser API over it to hide the specific internal representation, or just use tinyXml2 to sniff around in it.

The idea would be to parse `.urdf` and `.sdf` into an sdf-compatible Xml document. Do you think that is possible and makes sense?

liangfok, 2016-10-26:
Yes, I think it's possible and makes sense, but perhaps as a second spiral. This is because simply consolidating the parsers that are currently in `RigidBodySystem` and `parser_xxx` into `parser_xxx` will be a good first spiral since that'll at least physically decouple the parsing code from the simulation object code. We can then work on modifying the parsers to emit a TinyXML2 representation of the data, which the simulation objects consume to create themselves in a future spiral, assuming we decide to go that route.

jwnimmer-tri, 2016-10-26:
I think tinyXml2 would be an awful IR.  Have you looked at its APIs?  Yuck!  Half of the crappiness of the current parsing code comes from using tinyXml in the first place.  If XML is your one true god, then something like http://www.codesynthesis.com/products/xsd/ would be vastly superior to tinyXml.

liangfok, 2016-10-26:
Agreed. In addition to API issues, another problem is it does not provide line numbers, which makes it difficult to provide useful error messages when problems occur while parsing the XML.

sherm1, 2016-10-26:
> I think tinyXml2 would be an awful IR. 

I know it has an awful API. But there is so much benefit to having an IR I'd take anything over nothing! A bespoke thin accessor API in the parser for the few objects of interest could hide the IR details. 

XSD looks slick, but I've had bad luck with heavyweight XML libraries before (Xerces in particular). OTOH, I've been very happy using a nice API hiding an ugly TinyXML implementation.

jwnimmer-tri, 2016-10-26:
I guess I misunderstood the proposal.  I thought it was to use the tinyXml APIs.

sherm1, 2016-10-27:
> I guess I misunderstood the proposal. I thought it was to use the tinyXml APIs.

No, my bad. I was getting desperate so did say
> We could wrap a thin parser API over it to hide the specific internal representation, *or just use tinyXml2 to sniff around in it*.

I'm sorry! I take it back!

amcastro-tri, 2016-10-27:
I really like the idea of using some xml representation as our IR. I do agree with @sherm1 on that heavy lifting xml parsing libraries end up being a headache (I also used xerces in the past, bad decision).
What I did you in the past that offered all the capabilities that these heavy lifting libraries promise is `expat`. Expat is most likely even installed in your system already, it's super standard and stable. It offers a very basic but powerful API based on a few callbacks. Anyways, just something that worked for me in the past.

Something we talked with @liangfok and we think needs to be solved at some point is how to emit error messages when parsing. There are several levels of error messages: 1) From a bad xml formatting, 2) from a bad/incomplete user specification of drake specific entries, 3) from badly supplied parameters. How/when to report these in a nice stack-trace showing location in the file?

liangfok, 2016-10-27:
Bad XML formatting can probably be handled via the parser's fail-fast semantics, e.g,. the parser aborts immediately upon encountering it.

The rest of the failure modes require more finesse. Some can be detected while parsing an individual file. Others can only be detected after the parsing of all files and programmatic tweaks are complete, like during a "build simulation world" phase.

sherm1, 2016-10-27:
Producing nice error messages from parsing is likely to be an ongoing effort -- a first cut could be just to drop dead on anything bad. Subsequent refinement could be independent of the parse-to-a-simple-IR step.

liangfok, 2017-04-12:
With #4337 and the removal of `RigidBodySystem` from `master`, I think it's safe to close this issue. Any remaining sensor-specific parsing can be added as needed.

closed 2017-04-12 by liangfok