RobotLocomotion/drake #3496: bad nlopt library reference

RussTedrake, 2016-09-18:
i've been unable to actually run any exe's on my mac which touch drakeOptimization when nlopt is built.  

The symptom is
```
drake008% drake/bin/acrobotURDFDynamicsTest
dyld: Library not loaded: /Users/russt/drake-distro/build/lib/libnlopt.0.dylib
  Referenced from: /Users/russt/drake-distro/build/drake/bin/acrobotURDFDynamicsTest
  Reason: image not found
zsh: trace trap  drake/bin/acrobotURDFDynamicsTest
```
where somehow it looks for nlopt in `build/lib` instead of `build/install/lib`, as confirmed by `otool`
```
drake008% otool -L drake/bin/acrobotURDFDynamicsTest
drake/bin/acrobotURDFDynamicsTest:
	@rpath/libdrakeRBSystem.dylib (compatibility version 0.0.0, current version 0.0.0)
	/Users/russt/drake-distro/build/install/lib/libgtest.dylib (compatibility version 0.0.0, current version 0.0.0)
	/Users/russt/drake-distro/build/install/lib/libgtest_main.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libdrakeOptimization.dylib (compatibility version 0.0.0, current version 0.0.0)
	/Users/russt/drake-distro/build/lib/libnlopt.0.dylib (compatibility version 10.0.0, current version 10.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
	/Users/russt/drake-distro/build/install/lib/libgurobi60.so (compatibility version 6.0.0, current version 6.0.5)
	/Users/russt/drake-distro/build/install/lib/libmosek64.7.1.dylib (compatibility version 0.0.0, current version 7.1.45)
...
```
(note that gurobi and mosek, etc are in the right place)

but the pkg-config looks fine:
```
drake008% more nlopt.pc
prefix=/Users/russt/drake-distro/build/install
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=/Users/russt/drake-distro/build/install/include/nlopt

Name: NLopt
Description: nonlinear optimization libary
Version: 2.5
Libs: -L${libdir} -lnlopt -lm
Cflags: -I${includedir}
```

and in fact, the build command actually looks aok to me:
```
drake/examples/Acrobot/test/CMakeFiles/acrobotURDFDynamicsTest.dir/link.txt:/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++    -Werror=all -Werror=ignored-qualifiers -Werror=shadow -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -O3 -DNDEBUG -Wl,-search_paths_first -Wl,-headerpad_max_install_names  CMakeFiles/acrobotURDFDynamicsTest.dir/urdfDynamicsTest.cpp.o  -o ../../../bin/acrobotURDFDynamicsTest  -L/Users/russt/drake-distro/build/drake/lib  -L/Users/russt/drake-distro/build/install/lib ../../../lib/libdrakeRBSystem.dylib /Users/russt/drake-distro/build/install/lib/libgtest.dylib /Users/russt/drake-distro/build/install/lib/libgtest_main.dylib ../../../lib/libdrakeOptimization.dylib -L/Users/russt/drake-distro/build/install/lib -lnlopt -L/Users/russt/drake-distro/build/install/lib/dreal -lm -ldreal -libex -lClpSolver -lClp -lCoinUtils -lcapd -ladept -lglpk -lprim -lpicosat -lm -ldreal -libex -lClpSolver -lClp -lCoinUtils -lcapd -ladept -lglpk -lprim -lpicosat -lsnopt_c -lf2c -lgurobi60 -lmosek64 -liomp5 ../../../lib/libdrakeRigidBodyConstraint.dylib ../../../lib/libdrakeRBM.dylib ../../../lib/libdrakeCollision.dylib /Users/russt/drake-distro/build/install/lib/libBulletDynamics.dylib /Users/russt/drake-distro/build/install/lib/libBulletCollision.dylib /Users/russt/drake-distro/build/install/lib/libLinearMath.dylib /Users/russt/drake-distro/build/install/lib/libBulletSoftBody.dylib ../../../lib/libdrakeShapes.dylib ../../../lib/libdrakeJoints.dylib ../../../lib/libdrakeUtil.dylib ../../../lib/libdrakeXMLUtil.dylib ../../../lib/libtinyxml2.dylib ../../../lib/libspruce.dylib ../../../lib/libdrakeGeometryUtil.dylib ../../../lib/libdrakeMath.dylib ../../../lib/libdrakeCommon.dylib /Users/russt/drake-distro/build/install/lib/libgflags.2.2.0.dylib -Wl,-rpath,/Users/russt/drake-distro/build/drake/lib -Wl,-rpath,/Users/russt/drake-distro/build/install/lib
```

I'm not sure where the disconnect is happening.  any ideas?  @RobotLocomotion/kitware ?
i'm afraid i'm blocked on this (at least on this machine).

RussTedrake, 2016-09-18:
it starts back here
```
drake008% otool -L drake/lib/libdrakeOptimization.dylib
drake/lib/libdrakeOptimization.dylib:
	@rpath/libdrakeOptimization.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libdrakeCommon.dylib (compatibility version 0.0.0, current version 0.0.0)
	/Users/russt/drake-distro/build/lib/libnlopt.0.dylib (compatibility version 10.0.0, current version 10.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
	/Users/russt/drake-distro/build/install/lib/libgurobi60.so (compatibility version 6.0.0, current version 6.0.5)
	/Users/russt/drake-distro/build/install/lib/libmosek64.7.1.dylib (compatibility version 0.0.0, current version 7.1.45)
	/Users/russt/drake-distro/build/install/lib/libiomp5.dylib (compatibility version 5.0.0, current version 5.0.0)
	@rpath/libgflags.2.2.dylib (compatibility version 2.2.0, current version 2.2.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.1.0)
```

RussTedrake, 2016-09-18:
Progress.  Just found this:

```
[ 95%] Performing install step for 'nlopt'
Making install in util
Making install in direct
Making install in cdirect
Making install in praxis
Making install in luksan
Making install in crs
Making install in mlsl
Making install in mma
Making install in cobyla
Making install in newuoa
Making install in neldermead
Making install in auglag
Making install in bobyqa
Making install in isres
Making install in slsqp
Making install in esch
Making install in api
 /usr/local/bin/gmkdir -p '/Users/russt/drake-distro/build/install/include/nlopt'
 /usr/local/bin/ginstall -c -m 644 nlopt.h nlopt.f nlopt.hpp '/Users/russt/drake-distro/build/install/include/nlopt'
 /usr/local/bin/gmkdir -p '/Users/russt/drake-distro/build/install/share/man/man3'
 /usr/local/bin/ginstall -c -m 644 nlopt.3 '/Users/russt/drake-distro/build/install/share/man/man3'
Making install in .
 /usr/local/bin/gmkdir -p '/Users/russt/drake-distro/build/install/lib'
 /bin/sh ./libtool   --mode=install /usr/local/bin/ginstall -c   libnlopt.la '/Users/russt/drake-distro/build/install/lib'
libtool: install: /usr/local/bin/ginstall -c .libs/libnlopt.0.dylib /Users/russt/drake-distro/build/install/lib/libnlopt.0.dylib
libtool: install: (cd /Users/russt/drake-distro/build/install/lib && { ln -s -f libnlopt.0.dylib libnlopt.dylib || { rm -f libnlopt.dylib && ln -s libnlopt.0.dylib libnlopt.dylib; }; })
libtool: install: /usr/local/bin/ginstall -c .libs/libnlopt.lai /Users/russt/drake-distro/build/install/lib/libnlopt.la
libtool: install: /usr/local/bin/ginstall -c .libs/libnlopt.a /Users/russt/drake-distro/build/install/lib/libnlopt.a
libtool: install: chmod 644 /Users/russt/drake-distro/build/install/lib/libnlopt.a
libtool: install: ranlib /Users/russt/drake-distro/build/install/lib/libnlopt.a
libtool: warning: remember to run 'libtool --finish /Users/russt/drake-distro/build/lib'
 /usr/local/bin/gmkdir -p '/Users/russt/drake-distro/build/install/lib/pkgconfig'
 /usr/local/bin/ginstall -c -m 644 nlopt.pc '/Users/russt/drake-distro/build/install/lib/pkgconfig'
Making install in octave
Making install in test
Making install in swig
[ 95%] Completed 'nlopt'
```

notice the line about running `libtool —finish` on `build/lib`

RussTedrake, 2016-09-18:
and `drake/externals/nlopt/libnlopt.la` ends with
```
# Directory that this library needs to be installed in:
libdir='/Users/russt/drake-distro/build/lib'
```

i have convinced myself that we are not passing an incorrect directory in from cmake.

passing in exec_prefix explicitly seems to resolve the issue.  PR coming.

naveenoid, 2016-09-19:
BTW Im experiencing identical problems as well. Not sure why CI didnt catch it.

liangfok, 2016-09-19:
I just built Drake from a fresh clone on my OS X machine and was _unable_ to reproduce this error.

```
$ cd drake-distro
$ ./drake/bin/acrobotURDFDynamicsTest
Running main() from gtest_main.cc
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from AcrobotDynamicsTest
[ RUN      ] AcrobotDynamicsTest.ValueAssignment
[       OK ] AcrobotDynamicsTest.ValueAssignment (647 ms)
[----------] 1 test from AcrobotDynamicsTest (647 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (647 ms total)
[  PASSED  ] 1 test.
```
Maybe my machine could be used to determine why CI is not catching it?

jamiesnape, 2016-09-19:
I am unable to reproduce either.