RobotLocomotion/drake #3500: QuadraticErrorCost does not include the constant term

RussTedrake, 2016-09-19:
MathematicalProgram::AddQuadraticErrorCost creates a QuadraticConstraint, but ignores the constant term.  While this will not change the optimal solution, it violates the contract / documentation of the method and could lead to confusion if a user is trying to evaluate the objective (e.g. at the optimal solution)

http://drake.mit.edu/doxygen_cxx/classdrake_1_1solvers_1_1_mathematical_program.html#a49e093a81a903769209ef95cd6758424

hongkai-dai, 2016-12-30:
I tried to add a constant term in `QuadraticConstraint`, such that it means
```
lb <= 0.5 * x'*Q*x + b'*x + c <= ub
```
Unfortunately some solvers (e.g. Gurobi) does NOT accept a constant term in the objective function. That means the objective value computed by the solver will be different from calling `quadratic_cost.Eval(x)`, if there is a constant term.

RussTedrake, 2016-12-30:
thanks for looking at it.  could we potentially add it back in after the call to the solver?

hongkai-dai, 2016-12-30:
Yes, I believe that is the solution.

We do not have a method to get the solution from the solver directly. If we are going to implement this method (which I think we should), we need to be cautious that the objective value coming out from the solver (Gurobi) can be different from evaluating the cost by a constant term.

RussTedrake, 2017-04-29:
I just realized that this is especially bad now, because `AddCost(symbolic::Expression&)` keeps the constant term if the order is >2, but *drops* the constant term if the cost is linear or quadratic!  yuck!

@EricCousineau-TRI -- now that you've separated costs from constraints, perhaps you can address this one properly at last?

EricCousineau-TRI, 2017-04-30:
Sounds good!

EricCousineau-TRI, 2017-05-04:
Working on WIP. Will add tests into PR #6021 per Hongkai's request, and use that to test the constant offset.

EricCousineau-TRI, 2017-05-10:
Completed on EricCousineau-TRI/drake#5, but #6021 is a prerequisite. Will rebase once that is merged in.

closed 2017-07-17 by EricCousineau-TRI