RobotLocomotion/drake #3504: Call graph leaves are a semantically-depleted place to generate errors

david-german-tri, 2016-09-19:
The current error-generation conventions in Drake do not encourage propagating errors up the tree.  Instead of catching exceptions (forbidden) or returning error codes (permitted but not widely done), we tend to abort, assert, or throw an uncaught exception at the call graph leaf where the error is detected.

As a result, the information we emit on error tends to be sparse.  A lot of the context that would help the author find root cause is unavailable.  I personally cope with this by launching `gdb` at the drop of a hat.  Is that good enough?  Other possibilities include:

- Modifying the `drake_assert` macros to dump the stack.
- Evolving our code habits to include a lot more `spdlog` statements.
- Evolving our code habits to include more error codes.

jwnimmer-tri, 2016-09-19:
Note that the linked example to #3486 is specifically a throw (chosen, I believe, because it can be covered by unit testing), and not an assertion.

(1) We could make `drake::Abort` do a backtrace; that would help with `ASSERT` and `DEMAND` and is a good idea in general anyway.

(2) Getting `throw` to produce more context is a bit harder.  There are ways to "tag-forward" additional context (perhaps only in Debug mode?), but they require some infrastructure.

(3) See also `std::expected`.

But the specific case that motivated this, since a clear purpose of `CheckValidOutput` is to produce useful error message, I would suggest updating and/or rewriting that function to be as useful as possible with current tools.  Using something other than `get_vector_data` would solve this particular case.

jwnimmer-tri, 2016-09-19:
I'll also note that #2890 has a good chance of making this error message clearer just by itself.

david-german-tri, 2017-04-03:
I'm not sure there's any real action here.

closed 2017-04-03 by david-german-tri

jwnimmer-tri, 2017-05-24:
For the record, the term of art for (2) would be "error context stack".  Conceivably the system framework (in coordination with error-reporting in `//drake/common`) could eventually offer that.