RobotLocomotion/drake #3511: Can't build IRIS

tkoolen, 2016-09-19:
Building drake `WITH_ALL_SUPPORTED_EXTERNALS`, I'm getting the following error while building IRIS:
```
In file included from /Users/twan/code/drake-distro/externals/iris/iris/cvxgen/cvxgen_ldp.cpp:2:
/Users/twan/code/drake-distro/externals/iris/iris/cvxgen/cvxgen_ldp.h:8:10: fatal error: 'Eigen/Core' file not found
#include <Eigen/Core>
         ^
1 error generated.
```

Could possibly be related to https://github.com/RobotLocomotion/drake/pull/3439.

jamiesnape, 2016-09-19:
Honestly, no idea. The output of the configure seems to be suppressed. Neither `FindEigen.cmake` or `pkg-config` is good, though.

tkoolen, 2016-09-19:
It's strange. The eigen3.pc installed to drake-distro/build/install/share/pkgconfig looks good. The `-I` directory it refers to exists, and contains `Eigen/Core`. The only other eigen3.pc on my machine is the one in drake-distro/build/externals/eigen, which is identical.

tkoolen, 2016-09-20:
Talked to @rdeits about this, and we decided to just remove the `pods_find_pkg_config` stuff (i.e., always use `find_package( Eigen3 )`). I'll submit a PR to iris soon.

closed 2016-09-21 by david-german-tri