RobotLocomotion/drake #3512: Can't build dReal (link error related to log4cxx in CAPD4)

tkoolen, 2016-09-19:
Error occurs when building `WITH_ALL_SUPPORTED_EXTERNALS` on OSX.

CMake error:
```
[ 19%] Performing build step for 'CAPD4'
CMake Error at /Users/twan/code/drake-distro/build/externals/dreal/external/src/CAPD4-stamp/CAPD4-build-Release.cmake:16 (message):
  Command failed: 2

   'make'

  See also

    /Users/twan/code/drake-distro/build/externals/dreal/external/src/CAPD4-stamp/CAPD4-build-*.log
```

Contents of `CAPD4-build-err.log`:
```
Undefined symbols for architecture x86_64:
  "log4cxx::helpers::CharMessageBuffer::operator<<(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      capd::auxil::Logger::doInit() in libcapdauxil.a(libcapdauxil_la-Logger.o)
  "log4cxx::Logger::forcedLog(log4cxx::helpers::ObjectPtrT<log4cxx::Level> const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, log4cxx::spi::LocationInfo const&) const", referenced from:
      capd::auxil::Logger::doInit() in libcapdauxil.a(libcapdauxil_la-Logger.o)
      capd::auxil::Logger::forcedLogInfo(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, char const*, char const*, int) in libcapdauxil.a(libcapdauxil_la-Logger.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[10]: *** [BuildInfo] Error 1
make[9]: *** [all-recursive] Error 1
make[8]: *** [all] Error 2
make[7]: *** [all-recursive] Error 1
make[6]: *** [all-recursive] Error 1
```

I have `log4cxx` 0.10.0 installed through homebrew. It isn't explicitly listed as a dependency of drake or dReal, but is brought in by autoconf/automake/libtool.

tkoolen, 2016-09-19:
This appears to be resolved by running
```
brew rm log4cxx
brew install log4cxx --c++11 --universal
```
Closing.

closed 2016-09-19 by tkoolen

RussTedrake, 2016-09-19:
@soonho-tri -- if these are needed, they should be added to the setup instructions..

soonho-tri, 2016-09-19:
I will check.

soonho-tri, 2016-09-20:
@tkoolen, I'm removing `log4cxx` dependency from `CAPD4`. It should be an easy task by using `--without-log4cxx` configure option. But it doesn't work as I expected. Anyway I'll figure out a way. When I re-enable `WITH_DREAL` by default, I will make sure that this one is resolved.

tkoolen, 2016-09-20:
OK, that sounds good to me.

A-I-Ivanov, 2016-09-29:
Same issue exists in Ubuntu 16.04. I know this is in beta support. Please post the fix if you find it, and I'll do the same (if I manage to figure it out).

soonho-tri, 2016-09-29:
@A-I-Ivanov, it's fixed in the latest version of dReal. But drake is still pointing to an old version of dReal. For now, `WITH_DREAL` is `OFF` and you shouldn't have the problem unless you enable it manually.

I'll make a PR to bump dReal hash in drake soon anyway.

liangfok, 2016-10-05:
I suspect this issue may have returned in `mac-clang-ninja-nightly-matlab-debug`:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-clang-ninja-nightly-matlab-debug/58/console