RobotLocomotion/drake #3513: Compile errors in ptToPolyBullet_mex

tkoolen, 2016-09-19:
Likely due to https://github.com/RobotLocomotion/drake/pull/3472.

```
[ 41%] Building CXX object systems/trajectories/CMakeFiles/ptToPolyBullet_mex.dir/ptToPolyBullet_mex.cpp.o
/Users/twan/code/drake-distro/drake/systems/trajectories/ptToPolyBullet_mex.cpp:12:10: fatal error: 'LinearMath/btTransform.h' file not found
#include "LinearMath/btTransform.h"
         ^
/Users/twan/code/drake-distro/drake/systems/trajectories/replanFunnels_mex.cpp:48:10: fatal error: 'LinearMath/btTransform.h' file not found
#include "LinearMath/btTransform.h"
         ^
1 error generated.
make[5]: *** [systems/trajectories/CMakeFiles/replanFunnels_mex.dir/replanFunnels_mex.cpp.o] Error 1
make[4]: *** [systems/trajectories/CMakeFiles/replanFunnels_mex.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
1 error generated.
make[5]: *** [systems/trajectories/CMakeFiles/ptToPolyBullet_mex.dir/ptToPolyBullet_mex.cpp.o] Error 1
make[4]: *** [systems/trajectories/CMakeFiles/ptToPolyBullet_mex.dir/all] Error 2
/Users/twan/code/drake-distro/drake/systems/trajectories/shiftFunnel_snopt_mex.cpp:38:10: fatal error: 'LinearMath/btTransform.h' file not found
#include "LinearMath/btTransform.h"
         ^
1 error generated.
make[5]: *** [systems/trajectories/CMakeFiles/shiftFunnel_snopt_mex.dir/shiftFunnel_snopt_mex.cpp.o] Error 1
make[4]: *** [systems/trajectories/CMakeFiles/shiftFunnel_snopt_mex.dir/all] Error 2

```

So far I've spent most of my day fixing build issues. Is this really getting run in CI?

tkoolen, 2016-09-19:
I suppose this is just missing a `target_include_directories` involving `BULLET_INCLUDE_DIRS`.

david-german-tri, 2016-09-19:
> Is this really getting run in CI?

Yes, this error has shown up in the Mac MATLAB nightly builds for Friday, Saturday, and Sunday.  We don't run Mac MATLAB builds on every PR because they're prohibitively resource-intensive.  So, there was really no opportunity before today to catch these problems.

https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=116516

tkoolen, 2016-09-19:
OK, thanks. I'll submit a PR that fixes this.

tkoolen, 2016-09-19:
Fixed by #3510.

closed 2016-09-19 by tkoolen

jwnimmer-tri, 2016-09-19:
Also FWIW, it wasn't really running in CI as much as it should, due to #2754.