RobotLocomotion/drake #3524: Linker error in CAPD4 (dReal)

tkoolen, 2016-09-20:
CMake error:
```
[ 20%] Performing build step for 'CAPD4'
CMake Error at /Users/twan/code/drake-distro/build/externals/dreal/external/src/CAPD4-stamp/CAPD4-build-Release.cmake:16 (message):
  Command failed: 2

   'make'

  See also

    /Users/twan/code/drake-distro/build/externals/dreal/external/src/CAPD4-stamp/CAPD4-build-*.log
```

Relevant part of `drake-distro/build/externals/dreal/external/src/CAPD4-stamp/CAPD4-build-err.log`:
```
Undefined symbols for architecture x86_64:
  "capd::diffIncl::DiffInclusion<capd::diffIncl::MultiMap<capd::map::Map<capd::vectalg::Matrix<capd::intervals::Interval<double, capd::rounding::DoubleRounding>, 0u, 0u> >, capd::map::Map<capd::vectalg::Matrix<capd::intervals::Interval<double, capd::rounding::DoubleRounding>, 0u, 0u> > >, capd::dynsys::Solver<capd::map::Map<capd::vectalg::Matrix<capd::intervals::Interval<double, capd::rounding::DoubleRounding>, 0u, 0u> >, capd::dynsys::IEncFoundStepControl, capd::diffAlgebra::Curve<capd::diffAlgebra::BasicCurve<capd::vectalg::Matrix<capd::intervals::Interval<double, capd::rounding::DoubleRounding>, 0u, 0u> >, true> > >::~DiffInclusion()", referenced from:
      RosslerExample() in diffInclExample-diffInclExample.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

Platform: OSX/Clang.

tkoolen, 2016-09-20:
sha: 356bfb6916dccec51da6b900fc08dc5e228e33c8

soonho-tri, 2016-09-20:
@tkoolen, this is the side-effect of #3512. It should work now.

closed 2016-09-20 by soonho-tri

tkoolen, 2016-09-20:
Confirmed that this is resolved.

tkoolen, 2016-09-20:
Thanks for the quick response!