RobotLocomotion/drake #3535: Drake installation_Ubuntu_

chang-liang, 2016-09-20:
I followed http://drake.mit.edu/from_source.html  
and
 the Mandatory platform specific instructions(http://drake.mit.edu/ubuntu_trusty.html ).  
but after i executed "cmake .." in the step of "Build with Make" and waited for about an hour, the ubuntu terminal complained as follows (matlab version: MATLAB R2016a)

[image]

[image]

I still have the similar problem when building with Matlab version R2014a,
[image]

[image]

[image]

[image]

Please help.

tkoolen, 2016-09-21:
Could you try running `mex -setup` in Matlab, and then do a clean build?

tkoolen, 2016-09-21:
2016a works by the way, no need to downgrade Matlab.

tkoolen, 2016-09-21:
Also, for the `mex -setup` step I described above, be sure to start Matlab from the command line so that your `.bashrc` gets run and the `CC`, `CXX`, and `FC` environment variables you should be exporting there will be visible to Matlab at the time you run `mex -setup`.

chang-liang, 2016-09-21:
After i finished the instructions in http://drake.mit.edu/ubuntu_trusty.html
I opened MATLAB from ubuntu terminal, and matlab was opened. then i typed "mex -setup", it showed

[image]

then i chose " mex -setup C++ " in the matlab command window in the screen, it showed

[image]

chang-liang, 2016-09-21:
- It  shows it is not a successful setup, isn't it? 

- Does the problem caused probably by the installation order? i mean i installed matlab first and then did the platform-specific setup steps (http://drake.mit.edu/ubuntu_trusty.html)

- Should i go on to do a clean build?

chang-liang, 2016-09-21:
- It  shows it is not a successful setup, is it?

- Does the problem caused probably by the installation order? i mean i installed matlab first and then did the platform-specific setup steps (http://drake.mit.edu/ubuntu_trusty.html)

- Should i go on to do a clean build?

tkoolen, 2016-09-21:
Yeah, that's not good. I think Matlab may be explicitly looking for the `g++` program, instead of at the environment variables. Can you post the output of running
```bash
g++ -v
```
in the terminal?

Also, can you post the output of running
```matlab
!which g++
```
in Matlab?

chang-liang, 2016-09-21:
`g++ -v`
in the terminal

[image]

`g++ -v`
in the terminal agin

[image]

`!which g++` 
in Matalb

[image]

tkoolen, 2016-09-21:
OK, good, that's progress. But Matlab will still be using the wrong version of g++ (4.8 instead of 4.9). I think the instructions at http://drake.mit.edu/ubuntu_trusty.html may not be sufficient to properly set up a Matlab build then.

Please try running the following in your terminal:
```bash
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-4.9 60 --slave /usr/bin/g++ g++ /usr/bin/g++-4.9 
sudo update-alternatives --config gcc
```
and then select version 4.9 from the menu.

Verify that `g++` now points to `g++-4.9` by running `g++ -v` again (inspect the last line of the output).

Finally, rerun `mex -setup` in Matlab.

It is important to note that this changes your compiler *globally*. If you want to switch back to a different version of `gcc`/`/g++`, run `sudo update-alternatives --config gcc` again and select the version you want.

rpoyner-tri, 2016-09-21:
using update-alternatives for gcc will break your ubuntu installation. packages all over the place depend on the fact of 'gcc' being the distro default version.

rpoyner-tri, 2016-09-21:
In particular, trying to put gcc under update-alternatives will break DKMS. This will prevent you from installing packages that incorporate kernel modules, or (if you are unlucky), create unsafe/incompatible kernel modules.

tkoolen, 2016-09-21:
@rpoyner-tri, I'll hand it off to you. What exact steps would you recommend?

rpoyner-tri, 2016-09-21:
I can neither use nor fix Matlab. I am just trying to avoid recommending practices that are sure to break Linux distros.

@jhoare: are you willing to step in?

tkoolen, 2016-09-21:
We came up with a safer approach.

First, create a directory with symlinks to gcc-4.9 and g++-4.9:
```
cd ~
mkdir bin
cd bin
ln -s /usr/bin/gcc-4.9 gcc
ln -s /usr/bin/g++-4.9 g++
```
Then, open your .bashrc:
```
gedit ~/.bashrc
```
and add the following line at the end:
```
export PATH=$HOME/bin:$PATH
```
Save and close, then run
```
source ~/.bashrc
```

Verify that `g++` now points to `g++-4.9` by running `g++ -v` again (inspect the last line of the output).

Finally, open a new instance of matlab (run from the command line) and rerun `mex -setup`.

This still changes your compiler *globally*, but only for $USER, not for root, making it safer. If you want to switch back to the old version of `gcc`/`/g++`, comment out the line you added to .bashrc.

After all that, you'll have to do a clean build, sorry.

jamiesnape, 2016-09-22:
None of this will be an issue when we switch over the the `matlab_add_mex` functions in `FindMatlab`, which will be coming soon.

chang-liang, 2016-09-28:
Thank you all very much for your help, but i still got the same respond after running [mex -setup](url).

chang-liang, 2016-09-28:
Thank you all very much for your help, but i still got the same respond after running `mex -setup`.

tkoolen, 2016-09-29:
`mex -setup` should indeed result in the same response, but running `!which g++` in matlab should now point to the g++ in `~/bin`, which is a symlink to `g++-4.9`. If this is indeed the case, you're ready to do a clean drake build at this time.

chang-liang, 2016-09-29:
Thank you. That was progress!

[image]

[image]

But when I tested the installation, I was faced with this,

[image]

please help...

tkoolen, 2016-09-30:
The spotless build seems to get stuck at the matlab command line for you, not sure why yet (I don't see any errors). Try entering `exit` when the build process gets to this point and see if the build continues?

You could also consider just disabling building spotless for now by `cd`ing to your build directory, then running `ccmake .` (notice the additional c), then setting `WITH_SPOTLESS` to `OFF` (press enter), then `c` to configure, then `g` to generate and exit. Finally, run `make` in your build directory again (there should be no need to do a clean build).

tkoolen, 2016-09-30:
I see you're running Matlab 2016b. I personally haven't tested that version yet.

chang-liang, 2016-10-04:
I had all this Drake installation problems when I was using VMware (virtual machine) to run Ubuntu. Now I have set up a "real" Ubuntu in my PC hardware and I seems I have Drake installed successfully. Thank you very much to Twan, Rich and Jamie!

[image]

chang-liang, 2016-10-04:
I just skipped the part in "Using Gurobi as an external" in [http://drake.mit.edu/from_source.html](url). After the Drake installation, I found 'home/XXXX/drake-distro/externals/gurobi' is empty. Should I download Gurobi by myself? 

Actually I did. I put the downloaded Gurobi in 'drake-distro/externals/gurobi/', and got the license key

`cd drake-distro
./externals/gurobi/gurobi604/linux64/bin/grbgetkey [XXXXkey-numberXXXXX]`

But matlab still complains error like

"
Error using checkDependency (line 464)
Cannot find required dependency: gurobi

Error in runAtlasWalking (line 7)
checkDependency('gurobi');
"

Or should I not skip the part in "Using Gurobi as an external" in [http://drake.mit.edu/from_source.html](url), then where is ` WITH_ALL_SUPPORTED_EXTERNALS`, please? I didn't see

`Enter your password for gurobi.com:` 
(as you guys mentioned in in the instructions ) in the terminal during making Drake.

Any suggestion, please? Thank you very much!

PS: Environments:
Ubuntu 14.04 LTS (not run in the VMware virtual machine);
Matlab R2016a;
gcc 4.9;
g++ 4.9;

chang-liang, 2016-10-06:
What is the step of  building Drake with Gurobi included?
Could you please take some time to answer? I will appreciate a lot. @tkoolen

tkoolen, 2016-10-06:
Sorry about the response time. The easiest way is to install Gurobi as a Drake external (instead of separately).

In the `build` directory, run `ccmake .`. This will give you a menu where you can select the externals you want to download. Here you can toggle `WITH_GUROBI` to on, then press `c` to configure, then `g` to generate and quit. Then `make` again (there should be no need for a clean build). Note that quite a few additional targets in Drake will be built if Gurobi is present. You'll be prompted for your Gurobi email and password during the build process.

You can find the `grbgetkey` program, needed to download the Gurobi key file, in drake-distro/externals/gurobi/gurobi605/$ARCH/bin, although I guess you probably already ran that from your other Gurobi install by now.

Note that some of the externals that are enabled when using `WITH_ALL_SUPPORTED_EXTERNALS` require access to private repositories. `WITH_ALL_PUBLIC_EXTERNALS` enables only the publicly accessible ones. If all you need is Gurobi, just toggle `WITH_GUROBI`.

chang-liang, 2016-10-06:
Thank you, Twan! Since Drake is under construction, does a clean build of Drake will be affected for now and in the short future? @tkoolen

tkoolen, 2016-10-06:
You're very welcome. I'm not sure what you mean, but if any additional issues arise in the future, feel free to open a new issue.

closed 2016-10-06 by tkoolen

chang-liang, 2016-10-06:
You updated your web page [http://drake.mit.edu/index.html](url), adding "important note((October, 2016))" which says Drake is currently undergoing a major renovation. My question is whether there will be a problem if I do a clean build now.

tkoolen, 2016-10-06:
Oh no, not at all. *You* decide when you want to pull in the latest changes from master. Unless you do a `git pull`, you'll be compiling the exact same code every time (including the externals, which point to fixed commit SHAs), also if you do a clean build. BTW, I was saying it should **not** be necessary to do a clean build.

Staying up to date by following http://drake.mit.edu/from_source.html#stay-up-to-date will also be fine even during the renovation, since we have a fairly large amount of tests that are run whenever code gets changed. Really, the major renovation has been going on for quite a while; what's new is the notice on the website.