RobotLocomotion/drake #3537: drake-jenkins-bot spams PRs from users who are not org members

david-german-tri, 2016-09-21:
"Can one of the admins verify this patch?" ad infinitum.  Most recently #3536 (though apparently someone cleaned up the thread afterwards).

jamiesnape, 2016-09-21:
Very, very difficult to fix. I will try again, but it is a fundamental problem with the design of the GitHub pull request plugin.

david-german-tri, 2016-10-06:
Closing as will-not-fix; let's just make sure to add new contributors to the org.

jwnimmer-tri, 2017-01-15:
@david-german-tri Recent evidence suggests that this failure mode is really no way to run an open-source project.  Can we put this back on the radar, even if it doesn't happen right away?

If nothing else, it seems like forking the plug-in and commenting out the "verify this patch" emission should be a work-around.  I think having the bot post 0 this message times is way better than dozens of times.

jwnimmer-tri, 2017-03-30:
@jamiesnape @BetsyMcPhail How feasible is it to get _any_ kind of work-around in place for this?  Is it in fact possible to just turn off that one message entirely (or maybe all post-backs entirely)?

jamiesnape, 2017-03-30:
Someone would need to wade through the spaghetti of code in the plugin, but I think we are currently all backed up with working on the CMake-Bazel changeover, so not sure we have any free cycles to work on it immediately unless you would wish to reprioritize.

jwnimmer-tri, 2017-03-30:
Can you link to the plugin source code?  I bet @stonier could do it quickly.

jamiesnape, 2017-03-30:
Here, https://github.com/jenkinsci/ghprb-plugin, but I have to test it before it is deployed on production, and that takes time. That plugin is fundamental to our workflow, so unless this is suddenly a critical issue, this should wait.

jwnimmer-tri, 2017-03-30:
Alright, maybe we can prepare a candidate change but wait on testing / deploying.

jamiesnape, 2017-03-30:
Sure.

jwnimmer-tri, 2018-06-09:
I read through the plug-in's code.  In the Jenkins `/configure` screen, in the "GitHub Pull Request Builder" section, after opening the "Advanced" fold (near the "Admin List" textbox), there is a config setting named "Request for testing phrase".  I believe if we set that to empty, the bot will no longer spam us.

https://github.com/jenkinsci/ghprb-plugin/blob/b49fbc7/src/main/java/org/jenkinsci/plugins/ghprb/GhprbRepository.java#L243

@jamiesnape Can we try this, at some point?

jamiesnape, 2018-06-11:
We can, but unfortunately I have to go through every single job and update the setting, so it may take a while.

jwnimmer-tri, 2018-06-11:
I guess I was thinking about doing a quick test to confirm my reading of the code was correct, though yeah -- just changing one build's setting would be difficult to tell whether the spam went down by one.

jamiesnape, 2018-08-06:
I think this may finally be fixed. I just need a volunteer with a non-whitelisted GitHub account that does not mind the risk of getting spammed 84 times.

jamiesnape, 2018-08-07:
I guess I had volunteers in #9234 and #9236.

closed 2018-08-07 by jamiesnape