RobotLocomotion/drake #3543: Drake's IIWA Model Cannot be Loaded into Gazebo

liangfok, 2016-09-21:
# Problem Definition

[Drake's IIWA model](https://github.com/RobotLocomotion/drake/blob/a2cb01650579aa0e1ce4a4b19743fa641d7cdc10/drake/examples/kuka_iiwa_arm/urdf/iiwa14.urdf) currently references `.obj` (i.e., [Wavefront](https://en.wikipedia.org/wiki/Wavefront_.obj_file)) files because its visualizer [only supports this format](https://github.com/RobotLocomotion/drake/blob/5a7f663a2b9466c2b7225ad2ad1ebd0751aa5cac/drake/systems/plants/shapes/Geometry.cpp#L191). Unfortunately, Gazebo does _not_ support `.obj` files and instead supports `.dae` (i.e., [COLLADA](https://en.wikipedia.org/wiki/COLLADA)) files. Because of this, we are unable to load Drake's IIWA model into Gazebo. Being able to load the same model into Gazebo is beneficial since it will enable us to do comparisons and provide an alternative framework within which to debug and improve the model.

# Proposed Solutions

The following options are sorted from least effort the most effort.

## Option 1
Use [xacro](http://wiki.ros.org/xacro) to generate two URDFs. One will be called `iiwa14_drake.urdf`, which references `.obj` files, and other will be called `iiwa14_gazebo.urdf`, which references `.dae` files. Since the same code generator is used to create both files, there shouldn't be a problem keeping the two syncronized.

## Option 2
Convert `iiwa14.urdf` to [SDF] and use embedded ruby to generate a version for Gazebo and another version for Drake as described here: http://answers.gazebosim.org/question/14063/how-can-i-declare-a-variable-in-modelsdf-file/.

## Options 3
Modify Drake to support `.dae` files or Gazebo to support `.obj` files. I'm unsure how much work and time this will take.

# References
For reference, see: https://github.com/robotlocomotion/drake/pull/3507#issuecomment-248637867

sherm1, 2016-09-21:
Modifying Gazebo to support `.obj` files for geometry seems like the cleanest solution, and I think it would get used a lot. What do you think @nkoenig ?

nkoenig, 2016-09-21:
I agree with @sherm1. Gazebo should support `.obj` files. We'll get started on that asap.

RussTedrake, 2016-09-22:
Actually I believe director supports anything that vtk supports, include dae, etc.

nkoenig, 2016-10-06:
Gazebo 7 now supports loading OBJ files. A debian release will be available today or tomorrow.

sherm1, 2016-10-06:
This is great news, Nate -- thanks!

jwnimmer-tri, 2016-12-22:
Gazebo should be able to do this now / soon.

closed 2016-12-22 by jwnimmer-tri

liangfok, 2017-02-26:
@nkoenig, does Gazebo 8 not support `.obj`? I just tried loading Drake's IIWA model into Gazebo 8 and the visualizations did not work.