RobotLocomotion/drake #3544: Introduce DrakeNumTraits?

david-german-tri, 2016-09-21:
In the most straightforward (and likely most common) discrete system, the sampling rate is fixed.  Computing the next sample time in this case requires an integer rounding operation such as `ceil`.  Those operations will be available on many important scalar types, such as `double` and `AutoDiffScalar`, but not e.g. on `Polynomial`.  

We would like to be able to compile a simple discrete-time System specialized on any scalar type.  If the scalar type is incompatible with simulation, as `Polynomial` would be, code that instantiates that specialization and attempts to simulate it should get a runtime error.  To achieve this, we need conditional compilation that selects an appropriate implementation of `DoCalcNextSampleTime` for the scalar type.

`Eigen::NumTraits<T> is the [type trait struct for Eigen scalars] (https://eigen.tuxfamily.org/dox/NumTraits_8h_source.html).  None of the existing traits answer the question "does `ceil` make sense for this type"?  Therefore, I'm interested in adding a `DrakeNumTraits<T>` to `drake/common` for additional Drake-specific traits.  I think this first use case would be a boolean `IsEmbeddedInRealLine`, or something like that.

Any feedback on this idea before I go and do it?

sherm1, 2016-09-21:
`DrakeNumTraits` seems reasonable. My preference though would be to require all our Scalar types to be execution-compatible like AutoDiffScalar is. (We discussed that previously for FunctionalForm.) Without that numerical carry-along, the Scalar types can't follow the same execution path through the code as a plain double would do which limits their utility to straight-line code. Also, having them executable would avoid a lot of metaprogramming magic of the kind you are contemplating.

jwnimmer-tri, 2016-09-21:
> ... require all our Scalar types to be execution-compatible like AutoDiffScalar is.

I believe that this would make some kinds of analyses unsound.  Let's be sure to discuss before we choose that road.  (I don't think we need to pursue it immediately?)

RussTedrake, 2016-09-22:
fwiw - I think implementing our own NumTraits sounds completely reasonable (in fact, I can’t think of any reasonable alternative).

RussTedrake, 2016-09-22:
Matlab uses isNumeric instead of isOnRealLine.  I like that a bit better

amcastro-tri, 2016-09-30:
I completely support this!

david-german-tri, 2016-09-30:
It's in #3587!

PR #3587 (2016-10-04): Introduce discrete state for Systems
This includes both difference (numeric) and modal (arbitrary) state.  Add support to LeafSystem for fixed periodic state updates.  Add the ZeroOrderHold primitive as an example use case.

This commit does not add support to the simulator for actually updating discrete state.  It also does not add support to Diagrams for exposing the discrete state of their constituents.

There are a few interesting design points to discuss here.  +@edrumwri for feature review and +@sherm1 for platform review, but we should probably chat f2f before you dive deep into the cod [... cut]
closed 2016-10-04 by david-german-tri