RobotLocomotion/drake #3553: car_sim_lcm is broken

liangfok, 2016-09-21:
# The Error
The error is:

```
$ cd drake-distro/drake/automotive
$ ../../build/drake/bin/car_sim_lcm models/prius/prius.urdf models/stata_garage_p1.sdf
...
terminate called after throwing an instance of 'std::out_of_range'
  what():  Cannot set a BasicVector of size 3 with a value of size 0
Aborted (core dumped)
```

`gdb` reports a really messy backtrace due to templates:

```
gdb) bt
#0  0x00007ffff63ecc37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff63f0028 in __GI_abort () at abort.c:89
#2  0x00007ffff69fda9d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff69fba76 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff69fbac1 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff69fbd26 in __cxa_rethrow () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x000000000040a224 in __shared_count<drake::internal::LCMInputSystem<drake::CascadeSystem<drake::CascadeSystem<drake::Gain<drake::automotive::DrivingCommand1, drake::RigidBodySystem::StateVector>, drake::PDControlSystem<drake::RigidBodySystem> >, drake::BotVisualizer<drake::RigidBodySystem::StateVector> >::InputVector, void>, std::allocator<drake::internal::LCMInputSystem<drake::CascadeSystem<drake::CascadeSystem<drake::Gain<drake::automotive::DrivingCommand1, drake::RigidBodySystem::StateVector>, drake::PDControlSystem<drake::RigidBodySystem> >, drake::BotVisualizer<drake::RigidBodySystem::StateVector> >::InputVector, void> >, drake::CascadeSystem<drake::CascadeSystem<drake::Gain<drake::automotive::DrivingCommand1, drake::RigidBodySystem::StateVector>, drake::PDControlSystem<drake::RigidBodySystem> >, drake::BotVisualizer<drake::RigidBodySystem::StateVector> >&, std::shared_ptr<lcm::LCM>&> (__a=..., this=<synthetic pointer>) at /usr/include/c++/4.9/bits/shared_ptr_base.h:626
#7  __shared_ptr<std::allocator<drake::internal::LCMInputSystem<drake::CascadeSystem<drake::CascadeSystem<drake::Gain<drake::automotive::DrivingCommand1, drake::RigidBodySystem::StateVector>, drake::PDControlSystem<drake::RigidBodySystem> >, drake::BotVisualizer<drake::RigidBodySystem::StateVector> >::InputVector, void> >, drake::CascadeSystem<drake::CascadeSystem<drake::Gain<drake::automotive::DrivingCommand1, drake::RigidBodySystem::StateVector>, drake::PDControlSystem<drake::RigidBodySystem> >, drake::BotVisualizer<drake::RigidBodySystem::StateVector> >&, std::shared_ptr<lcm::LCM>&> (__a=..., __tag=..., 
    this=<synthetic pointer>) at /usr/include/c++/4.9/bits/shared_ptr_base.h:1090
#8  shared_ptr<std::allocator<drake::internal::LCMInputSystem<drake::CascadeSystem<drake::CascadeSystem<drake::Gain<drake::automotive::DrivingCommand1, drake::RigidBodySystem::StateVector>, drake::PDControlSystem<drake::RigidBodySystem> >, drake::BotVisualizer<drake::RigidBodySystem::StateVector> >::InputVector, void> >, drake::CascadeSystem<drake::CascadeSystem<drake::Gain<drake::automotive::DrivingCommand1, drake::RigidBodySystem::StateVector>, drake::PDControlSystem<drake::RigidBodySystem> >, drake::BotVisualizer<drake::RigidBodySystem::StateVector> >&, std::shared_ptr<lcm::LCM>&> (__a=..., __tag=..., 
    this=<synthetic pointer>) at /usr/include/c++/4.9/bits/shared_ptr.h:316
#9  allocate_shared<drake::internal::LCMInputSystem<drake::CascadeSystem<drake::CascadeSystem<drake::Gain<drake::automotive::DrivingCommand1, drake::RigidBodySystem::StateVector>, drake::PDControlSystem<drake::RigidBodySystem> >, drake::BotVisualizer<drake::RigidBodySystem::StateVector> >::InputVector, void>, std::allocator<drake::internal::LCMInputSystem<drake::CascadeSystem<drake::CascadeSystem<drake::Gain<drake::automotive::DrivingCommand1, drake::RigidBodySystem::StateVector>, drake::PDControlSystem<drake::RigidBodySystem> >, drake::BotVisualizer<drake::RigidBodySystem::StateVector> >::InputVector, void> >, drake::CascadeSystem<drake::CascadeSystem<drake::Gain<drake::automotive::DrivingCommand1, drake::RigidBodySystem::StateVector>, drake::PDControlSystem<drake::RigidBodySystem> >, drake::BotVisualizer<drake::RigidBodySystem::StateVector> >&, std::shared_ptr<lcm::LCM>&> (__a=...) at /usr/include/c++/4.9/bits/shared_ptr.h:588
#10 make_shared<drake::internal::LCMInputSystem<drake::CascadeSystem<drake::CascadeSystem<drake::Gain<drake::automotive::DrivingCommand1, drake::RigidBodySystem::StateVector>, drake::PDControlSystem<drake::RigidBodySystem> >, drake::BotVisualizer<drake::RigidBodySystem::StateVector> >::InputVector, void>, drake::CascadeSystem<drake::CascadeSystem<drake::Gain<drake::automotive::DrivingCommand1, drake::RigidBodySystem::StateVector>, drake::PDControlSystem<drake::RigidBodySystem> >, drake::BotVisualizer<drake::RigidBodySystem::StateVector> >&, std::shared_ptr<lcm::LCM>&> () at /usr/include/c++/4.9/bits/shared_ptr.h:604
#11 drake::runLCM<drake::CascadeSystem<drake::CascadeSystem<drake::Gain<drake::automotive::DrivingCommand1, drake::RigidBodySystem::StateVector>, drake::PDControlSystem<drake::RigidBodySystem> >, drake::BotVisualizer<drake::RigidBodySystem::StateVector> > > (sys=std::shared_ptr (count 2, weak 0) 0x6700c0, lcm=std::shared_ptr (count 3, weak 0) 0x62a940, t0=t0@entry=0, tf=tf@entry=inf, x0=..., options=...)
    at /home/liang/dev/drake_catkin_workspace/src/drake/drake/../drake/systems/LCMSystem.h:237
#12 0x0000000000403b39 in drake::automotive::(anonymous namespace)::do_main (argc=<optimized out>, argv=<optimized out>) at /home/liang/dev/drake_catkin_workspace/src/drake/drake/automotive/car_sim_lcm.cc:50
#13 0x00007ffff63d7f45 in __libc_start_main (main=0x4032a0 <main(int, char const**)>, argc=2, argv=0x7fffffffd6b8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, 
    stack_end=0x7fffffffd6a8) at libc-start.c:287
#14 0x0000000000403307 in _start ()
```

# SHA
The above error was obtained using d26808c14c353a5f26d679ce110751dcb6f6ff5a.

liangfok, 2016-09-22:
I'm starting to look into this issue now.

liangfok, 2016-09-22:
The problem occurs during the creation of a `drake::LCMInputSystem`.

liangfok, 2016-09-22:
The problem is further narrowed down to the initializer list of `drake::LCMInputSystem`:

https://github.com/RobotLocomotion/drake/blob/2abef6973d416d84ccc2ef16fcf09ba62544b774/drake/systems/LCMSystem.h#L83

liangfok, 2016-09-23:
I instrumented `drake::getNumInputs()` (see: https://github.com/liangfok/drake/commit/2cd16f35d84c7cb03401c1754514a72cec054f39) and obtained the following output:

```
$ ../../build/drake/bin/car_sim_lcm models/prius/prius.urdf models/stata_garage_p1.sdf
getNumInputs(): Method called!
getNumInputs(): Num inputs: 3
getNumInputs(): Method called!
getNumInputs(): Num inputs: 3
Creating LCM input system...
getNumInputs(): Method called!
getNumInputs(): Method called!
getNumInputs(): Method called!
getNumInputs(): Num inputs: 0
getNumInputs(): Num inputs: 0
getNumInputs(): Num inputs: 0
libc++abi.dylib: terminating with uncaught exception of type std::out_of_range: Cannot set a BasicVector of size 3 with a value of size 0
Abort trap: 6
```
Clearly, the problem is one of the systems in the system's cascade has zero inputs.

jwnimmer-tri, 2016-09-23:
Did you `git bisect` yet?

Perhaps the loss of `LCMMessageType` in this commit broke the System1 magic LCM detection?
https://github.com/RobotLocomotion/drake/pull/3519/commits/144905741842758bf0618738799a8cd245b2eb3d#diff-6d385112eddbb09037e198b79e10b976

liangfok, 2016-09-23:
Thanks for the advice. I have not tried `git bisect` yet. Will do so soon.

liangfok, 2016-09-23:
@jwnimmer-tri, you guessed right. 144905741842758bf0618738799a8cd245b2eb3d has the bug but the previous version, 4bef78357bba9c16384b85e1f5eab690c3908062, does not.

PR #3572 (2016-09-26): Fixed car_sim_lcm Demo
This required:

1. Making `DrivingCommand1` a class.
2. Adding `LCMMessageType` and `channel()` to the class.

Resolves #3553.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3572)
closed 2016-09-26 by ggould-tri