RobotLocomotion/drake #3556: Require that all children of a Diagram have unique names.

david-german-tri, 2016-09-21:


sherm1, 2016-09-22:
:+1: This would allow unique "pathnames" for each subsystem in a diagram. It could also be used to provide unique input and output port names and unique parameter names.

liangfok, 2016-09-22:
Just wondering: Does this apply to "immediate" children or all the way down to the youngest descendants? For example, if I create a `Diagram` called "wheel" that contains numerous systems like "Tire", "Rim", "Bearing", "Bushing", "HubCap", "ValveStem", can I add four instances of this `Diagram` to a "Vehicle" Diagram? Is the proposed rule "...to have unique full path names", which includes the name of the ancestor `Diagram` objects?

I was just looking at http://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_diagram.html#a628bfe0e8835a0144d79b2a38424420d and didn't see a name associated with a `Diagram`.

sherm1, 2016-09-22:
> Does this apply to "immediate" children or all the way down

Just immediate children -- it's exactly analogous to file names in a directory structure.

liangfok, 2016-09-23:
Thanks. I just realized that `Diagram` is a subclass of `System`, which has `get_name()` and `set_name()`.

david-german-tri, 2017-04-14:
There are several existing Diagrams that violate this rule.  I'm sizing up a cleanup pass now.

PR #5866 (2017-04-18): Require that each subsystem of a Diagram has a unique name.
Fixes #3556.

@sherm1 for feature review.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5866)
commit 477e86c3 (2017-04-18): Require that each subsystem of a Diagram has a unique non-empty name.

Fixes #3556.
closed 2017-04-18 by david-german-tri