RobotLocomotion/drake #3568: `CompareMatrices` almost works for integer scalar types, but doesn't

tkoolen, 2016-09-23:
A call to `CompareMatrices` with integer matrix arguments compiles on all platforms except win32, and I believe that visual studio is right to complain.

See https://drake-cdash.csail.mit.edu/index.php?project=Drake&showfilters=1&filtercount=2&showfilters=1&filtercombine=and&field1=label&compare1=61&value1=jenkins-windows-msvc-ninja-32-experimental-3567-debug-4938&field2=buildstarttime&compare2=84&value2=now for a build that errored.

Error:
```
C:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt\math.h(403): error C2668: 'fpclassify': ambiguous call to overloaded function

C:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt\math.h(288): note: could be 'int fpclassify(long double) throw()'

C:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt\math.h(283): note: or       'int fpclassify(double) throw()'

C:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt\math.h(278): note: or       'int fpclassify(float) throw()'

C:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt\math.h(403): note: while trying to match the argument list '(int)'

C:\Users\Administrator\workspace\experimental\0b4ef046\drake/common/eigen_matrix_compare.h(53): note: see reference to function template instantiation 'bool isnan<int>(_Ty) throw()' being compiled
```
Even though http://en.cppreference.com/w/cpp/numeric/math/isnan describes a 
```c++
bool isnan( Integral arg )
```
I believe the standard doesn't require the existence of a function with that signature. It only requires `isnan` functions taking `float`, `double`, or `long double` arguments (http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2014/n4296.pdf).

jwnimmer-tri, 2016-09-23:
I've always wanted to rewrite that function using a proper googletest comparator (I forget their term for it).  Perhaps this a good opportunity.

david-german-tri, 2016-09-23:
>proper googletest comparator (I forget their term for it)

Matcher: https://github.com/google/googletest/blob/master/googlemock/docs/CookBook.md#extending-google-mock

RussTedrake, 2016-09-25:
Feels like another case of msvc detecting a real issue that g++ and clang did not?

david-german-tri, 2016-09-25:
> Feels like another case of msvc detecting a real issue that g++ and clang did not?

I don't share @tkoolen's read of the standard.  26.8-11 does specify an `isnan(Integral arg)`, as cppreference describes.  Relevant quote, emphasis added:

> Moreover, there shall be additional overloads sufficient to ensure:

> 1. If any arithmetic argument corresponding to a double parameter has type long double, then all
arithmetic arguments corresponding to double parameters are effectively cast to long double.

> 2. Otherwise, if any arithmetic argument corresponding to a double parameter has type double **or an integer type**, then all arithmetic arguments corresponding to double parameters are effectively cast to double.

If MSVC doesn't provide those overloads, I think that's just another way in which it's not C++11 compliant.

tkoolen, 2016-09-26:
Yep, you're right, I missed that (as did Microsoft :-)).

david-german-tri, 2016-09-26:
> as did Microsoft

In their defense (or something), they never claimed to be C++11 compliant.

david-german-tri, 2016-10-06:
Closing, since Windows is no longer supported.

closed 2016-10-06 by david-german-tri