RobotLocomotion/drake #3571: CLion setup guide is GCC-specific

tkoolen, 2016-09-23:
The CLion setup guide at http://drake.mit.edu/clion.html contains:
>
Under CMake Options expand the tab Pass system and custom environment. Add the following environment variables. (You can copy these from this documentation one at a time and click on the paste button at the right of the environment variables dialog.)
CC=gcc-4.9
CXX=g++-4.9
FC=gfortran-4.9
F77=gfortran-4.9

A labmate who's just getting up to speed on Drake development and is using OSX/clang followed these instructions to the letter, resulting in cmake complaining about not being able to find a compiler.

naveenoid, 2016-10-03:
I had the same problem after update of my Clion version. Have you got any updates on this issue?

tkoolen, 2016-10-03:
Just don't define those variables.

jwnimmer-tri, 2016-10-03:
We just made that edit for Ubuntu 16.04 users to this same document.  Sounds like adding an OSX caveat along the same lines would be an easy fix.

RussTedrake, 2016-11-27:
@seancurtis-tri has updated the instructions recently and seems to have taken on the role of local clion guru.  assigning this to him (possibly to close as “done”)

SeanCurtis-TRI, 2016-11-28:
The documentation for this step says:

> [This step only for Ubuntu 14.04 - Trusty]. Under CMake Options expand...

I'm content that it should be sufficiently clear for mac users.  I'm closing this.

closed 2016-11-28 by SeanCurtis-TRI