RobotLocomotion/drake #3592: Mysterious CI Failure: MATLAB Killed because the Java VM Could not be Terminated

liangfok, 2016-09-26:
# The Error

```
MATLAB killed because the Java VM could not be terminated for 30 seconds.
This could be due to a thread or timer that started right before shutdown,
or a finalizer or shutdown hook that took too long to run.
```

# Example Logs

* https://drake-jenkins.csail.mit.edu/job/experimental/5033/compiler=gcc,label=linux_matlab,type=release/

david-german-tri, 2016-09-26:
Well, that's a new one.  I'm going to guess it's a flake that will be rarely reproduced and never explained.  @BetsyMcPhail @jamiesnape assigned to you in case you think otherwise, but feel free to close unresolved rather than investigate.

jamiesnape, 2016-09-26:
I would not know where to start to reproduce it. I doubt it is CI specific.

closed 2016-09-26 by jamiesnape

nemo8130, 2019-02-27:
Same Error ! 

MATLAB killed because the Java VM could not be terminated for 30 seconds.
This could be due to a thread or timer that started right before shutdown,
or a finalizer or shutdown hook that took too long to run.

I have a 4 GB RAM in Ubuntu 18.04