RobotLocomotion/drake #3594: "with_all_supported_externals" build fails in CLP inside dreal

rdeits, 2016-09-26:
On OSX, I'm getting the rather unhelpful error message:

```
[  0%] Performing build step for 'CLP'
CMake Error at /Users/rdeits/locomotion/drake-distro/build/externals/dreal/external/src/CLP-stamp/CLP-build-Release.cmake:16 (message):
  Command failed: 2

   'make'

  See also

    /Users/rdeits/locomotion/drake-distro/build/externals/dreal/external/src/CLP-stamp/CLP-build-*.log

make[5]: *** [external/src/CLP-stamp/CLP-build] Error 1
make[4]: *** [CMakeFiles/CLP.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [dreal-prefix/src/dreal-stamp/dreal-build] Error 2
make[1]: *** [CMakeFiles/dreal.dir/all] Error 2
make: *** [all] Error 2
```

The logfile mentioned is also not very helpful:

```
Making all in Data/Sample
Making all in Data/Netlib
Making all in CoinUtils
Making all in src
Making all in Osi
Making all in src/Osi
Making all in src/OsiCommonTest
Making all in Clp
Making all in src
 /usr/local/opt/ccache/libexec/c++ -DHAVE_CONFIG_H -I. -I/Users/rdeits/locomotion/drake-distro/build/externals/dreal/external/src/CLP/Clp/src -I/Users/rdeits/locomotion/drake-distro/build/externals/dreal/external/src/CLP/CoinUtils/src -I/Users/rdeits/locomotion/drake-distro/build/externals/dreal/external/src/CLP-build/CoinUtils/src -DCOIN_HAS_CLP --std=c++11 -DCLP_BUILD -MT Clp_C_Interface.lo -MD -MP -MF .deps/Clp_C_Interface.Tpo -c /Users/rdeits/locomotion/drake-distro/build/externals/dreal/external/src/CLP/Clp/src/Clp_C_Interface.cpp -o Clp_C_Interface.o
```

jwnimmer-tri, 2016-09-26:
The Jenkins https://drake-jenkins.csail.mit.edu/view/Continuous/job/mac-clang-continuous-release/ seems to be purring along.  I wonder what is different between Jenkins and the local build.

/CC @soonho-tri

rdeits, 2016-09-26:
in `drake-distro/build/externals/dreal/external/src/CLP-stamp/CLP-build-err.log` I see the following: 

```
/Users/rdeits/locomotion/drake-distro/build/externals/dreal/external/src/CLP/Clp/src/Clp_C_Interface.cpp:206:11: error: use of undeclared identifier 'CLP_VERSION'
   return CLP_VERSION;
          ^
/Users/rdeits/locomotion/drake-distro/build/externals/dreal/external/src/CLP/Clp/src/Clp_C_Interface.cpp:211:11: error: use of undeclared identifier 'CLP_VERSION_MAJOR'
   return CLP_VERSION_MAJOR;
          ^
/Users/rdeits/locomotion/drake-distro/build/externals/dreal/external/src/CLP/Clp/src/Clp_C_Interface.cpp:215:11: error: use of undeclared identifier 'CLP_VERSION_MINOR'
   return CLP_VERSION_MINOR;
          ^
/Users/rdeits/locomotion/drake-distro/build/externals/dreal/external/src/CLP/Clp/src/Clp_C_Interface.cpp:219:11: error: use of undeclared identifier 'CLP_VERSION_RELEASE'
   return CLP_VERSION_RELEASE;
          ^
4 errors generated.
make[3]: *** [Clp_C_Interface.lo] Error 1
make[2]: *** [all] Error 2
make[1]: *** [all-recursive] Error 1
make: *** [all-recursive] Error 1
```

soonho-tri, 2016-09-26:
@rdeits, I'm on it.

RussTedrake, 2016-09-27:
This is probably the same error i got on my work laptop 2 weeks ago.  (Still unresolved)

soonho-tri, 2016-09-27:
@rdeits and @RussTedrake : I'm switching from building CLP from source to using a pre-built package of CLP. In Ubuntu, there is `coinor-libclp-dev` package (>=12.04) so it's an easy task. I'm working on checking [a homebrew formula for CLP](https://github.com/coin-or-tools/homebrew-coinor/blob/master/clp.rb).

jwnimmer-tri, 2016-09-27:
Perhaps until dReal uses a repeatable build, we should elevate #3505 to high priority and fix it separate from other new features?

soonho-tri, 2016-09-27:
AFAIK, https://github.com/RobotLocomotion/drake/pull/3522 is waiting for platform review.

naveenoid, 2016-09-27:
BTW I get this error too..I dont build with DREAL at the moment.

soonho-tri, 2016-10-11:
This should be fixed now. If not, please contact me.

closed 2016-10-11 by soonho-tri