RobotLocomotion/drake #3596: updated cmake externals fail to find mex when pdftex is installed (on mac homebrew)

RussTedrake, 2016-09-27:
e.g.
```
[ 58%] Performing configure step for 'iris_project'
CMake Error at cmake/mex.cmake:36 (message):
  Could not find MEX_CC using mex -v
Call Stack (most recent call first):
  cmake/mex.cmake:156 (get_mex_option)
  CMakeLists.txt:9 (mex_setup)
```

the reason is
```
drake008% mex
This is pdfTeX, Version 3.14159265-2.6-1.40.16 (TeX Live 2015) (preloaded format=mex)
 restricted \write18 enabled.
```

this was a case that I had handled specifically before... making sure that we call the mex that is in the same path as the matlab executable, even if the mex executable is pre-empted by pdftex in the path.  @jamiesnape -- i believe this happened with all of your cmake external updates?  PTAL?

jamiesnape, 2016-09-27:
If it is happening during configure for `iris`, then it would be using an unchanged version of `mex.cmake` bundled with `iris`, but I will, of course, double check and fix.

jamiesnape, 2016-09-27:
(BTW this will not be an issue when we move to using the native CMake MEX support, which is high priority).

jamiesnape, 2016-09-27:
We have `NO_DEFAULT_PATH` here and `MATLAB_ROOT/bin` is a HINT: https://github.com/RobotLocomotion/drake/blob/master/cmake/mex.cmake#L100

Are there issues with externals other than iris?

RussTedrake, 2016-09-28:
Yes.  Snopt fails with the same message

jamiesnape, 2016-09-28:
Ok. I will look into it further now. Incidentally, RobotLocomotion/snopt#9 will fix the snopt failure regardless of the current cause.

jamiesnape, 2016-09-28:
You are using MacTeX 2015?

jamiesnape, 2016-09-28:
Seems to be a problem with the original `mex.cmake` files because `cmake/mex.cmake:156` is a comment on the version with included with drake.

jamiesnape, 2016-09-28:
Not sure why you are only seeing it now, though.

RussTedrake, 2016-10-02:
:facepalm:  this was due to http://drake.mit.edu/faq.html?highlight=mex#could-not-find-mex-cc-using-mex-v-on-mac-after-updating-xcode .  apologies.  i forgot i had just updated xcode on my home machine.

closed 2016-10-02 by RussTedrake