RobotLocomotion/drake #3608: DrakeJoints Contains Macros that Violate Style Guide

liangfok, 2016-09-28:
`drake-distro/drake/systems/plants/joints/DrakeJoint.h` contains a macro called `POSITION_AND_VELOCITY_DEPENDENT_METHODS` that violates our style guide.

From our [style guide](https://google.github.io/styleguide/cppguide.html#Preprocessor_Macros):

> Don't use macros to define pieces of a C++ API.

I encountered this issue when I tried to add Doxygen comment blocks for the methods that were defined via macro.

liangfok, 2016-10-17:
@sherm1, I'm thinking about resolving this issue by making the entire `DrakeJoint` class templated on the Eigen scalar type and adopting the [-inl.h](http://drake.mit.edu/cxx_inl.html#three-file-inl-h-pattern-approach) pattern. What do you think?

sherm1, 2016-10-17:
@liangfok, please don't modify this yet. We are planning to make major modifications to this code so that we can use virtual methods in abstract joints. Let's discuss with @amcastro-tri who will be back soon.

liangfok, 2016-10-18:
Great! I will wait.

amcastro-tri, 2016-10-21:
We finally decided with @sherm1 and @RussTedrake to go ahead and template on `<T>` the whole RBT. The biggest motivations being to support auto-differentiation with respect to parameters and lumped parameters estimation. That is part of my planned work for this quarter. As a consequence, this will also solve this issue and the Doxygen issue. Therefore, I will assign this to myself now.

liangfok, 2016-10-21:
Sounds good! Are you going to adopt the [-inl.h](http://drake.mit.edu/cxx_inl.html#three-file-inl-h-pattern-approach) pattern?

amcastro-tri, 2016-10-21:
Not sure about that yet. @sherm1, didn't you have at the time an alternative solution to this? I remember some code from you in `simulator.h` [here](https://github.com/RobotLocomotion/drake/blob/348e8f73df8b5ed25539c41cc7f8a2993419fc83/drake/systems/analysis/simulator.h#L239) but removed by @jwnimmer-tri's PR #3802.

jwnimmer-tri, 2016-10-21:
See #3741.

amcastro-tri, 2016-10-21:
Thank you @jwnimmer-tri! great read!. Still catching up after vacation.

jwnimmer-tri, 2018-12-21:
This issue is about the RBT code, in the attic.  We're not going to fix it, it's okay to leave it GSG-non-compliant.

closed 2018-12-21 by jwnimmer-tri