RobotLocomotion/drake #3610: "test" project does not appear in CLion

SeanCurtis-TRI, 2016-09-28:
The expectation is that there is a CMake target that should appear as a target in build called "test", which should invoke running all tests.  In my particular build (and, anecdotally, others) it is missing.

The full list of targets which CLion *does* recognize are included below as screen captures.

[image]
[image]
[image]
[image]

jamiesnape, 2017-11-15:
@SeanCurtis-TRI Is this still an issue for you. I am assuming probably not.

SeanCurtis-TRI, 2017-11-15:
Correct. We can close this out.

closed 2017-11-15 by SeanCurtis-TRI