RobotLocomotion/drake #3615: Drake/ROS CI Needs xacro.py To Be Installed

liangfok, 2016-09-29:
# Motivation

It is often useful to generate a URDF using [xacro](http://wiki.ros.org/xacro) since it enables small variations of a particular model to be easily kept in sync and reduces verbosity when a model consists of numerous similar parts.

PR #3609 includes a test that uses `xacro.py` to generate and load a URDF into the ROS parameter server. The apparent lack of `xacro.py` is causing that PR to fail CI testing.

# Action Item

Install `xacro.py` by executing the following command:

```
$ sudo apt-get install ros-indigo-xacro
```

liangfok, 2016-09-29:
@david-german-tri, I'm going to assign this to you. Can you please reassign it to the appropriate person? Thanks.

liangfok, 2016-09-29:
In https://github.com/RobotLocomotion/drake/issues/2369, `ros-indigo-desktop` and `ros-indigo-ros-base` are both mentioned as Debian packages to install.

Using `apt-cache showpkg [package-name]`, I see that:

(1) `ros-indigo-xacro` is included by `ros-indigo-robot`.

(2) `ros-indigo-robot` is included by `ros-indigo-desktop` and `ros-indigo-ros-base`.

Thus, I believe `xacro` should be installed on Drake's ROS CI images.

liangfok, 2016-09-29:
In https://github.com/RobotLocomotion/drake/issues/2369#issuecomment-220453697, @jamiesnape notes that the ROS environment is configured via a driver-script in `drake-ci` since shell startup scripts are not called by Jenkins. 

Here's where the ROS environment is configured: https://github.com/RobotLocomotion/drake-ci/blob/master/ctest_driver_script_catkin.cmake#L97. It looks right since it includes the following two lines, which are necessary to find `xacro`:

```
set(ENV{ROS_PACKAGE_PATH} "/opt/ros/indigo/share:/opt/ros/indigo/stacks")
set(ENV{PYTHONPATH} "/opt/ros/indigo/lib/python2.7/dist-packages:$ENV{PYTHONPATH}")
```

Based on this, I'm beginning to suspect that `ros-indigo-xacro` is not installed on the CI image.

liangfok, 2016-10-10:
@jamiesnape, do you know if Debian package `ros-indigo-xacro` is installed on the Drake+ROS CI images?

jamiesnape, 2016-10-10:
@liangfok I will check.

jamiesnape, 2016-10-10:
My notes say that we ran the following per #2369:
```
sudo apt-get install ros-indigo-ros-base ros-indigo-tf ros-indigo-ackermann-msgs 
```
I will shell into an EC2 instance and look for `xacro`.

liangfok, 2016-10-10:
Thanks. If it is not installed, can you install it using the following command?

```
$ sudo apt-get install ros-indigo-xacro
```

jamiesnape, 2016-10-10:
It was not installed. I updated the image.

liangfok, 2016-10-10:
Thanks!! I will close this issue.

closed 2016-10-10 by liangfok