RobotLocomotion/drake #3619: Implement Affine / Linear System in System 2.0

naveenoid, 2016-09-29:
There is a need to port over the existing Affine and Linear Systems to System2.0

liangfok, 2016-10-14:
@naveenoid, I'm going to need this to port `car_sim_lcm` to System 2.0. Have you started on this? If not, can I start working on it now?

liangfok, 2016-10-16:
@naveenoid and I chatted a few hours ago; he's nearing completion of System 2.0 `AffineSystem` and `LinearSystem`.

@naveenoid, here's my rough draft of `AffineSystem`. Maybe you can recycle parts of it, especially the unit test?
https://github.com/liangfok/drake/commit/09a76dbbf09e99dec799764a0c100c8353f7bc2b

naveenoid, 2016-10-16:
Integrating now.

liangfok, 2016-10-16:
Thanks! Might as well also port the Gain system. 

https://github.com/RobotLocomotion/drake/blob/master/drake/systems/LinearSystem.h#L179

naveenoid, 2016-10-27:
Fixed in #3828

closed 2016-10-27 by naveenoid