RobotLocomotion/drake #3625: Update Jenkins Interaction Documentation to Include Scheduling Via Github Comment

liangfok, 2016-09-29:
This is the page that should be updated:

http://drake.mit.edu/from_source_ros.html#scheduling-a-drake-ros-continuous-integration-test

The above-linked website only describes how CI tests can be scheduled through Jenkin's website.
In addition to this, we can also schedule tests via comments in Github that take the following form:

> @drake-jenkins-bot foo-experimental-bar please.

This issue tracks the addition of documentation about this capability.

jamiesnape, 2016-09-29:
> To schedule a full test of Drake + ROS + MATLAB with gcc, post the following comment on your PR:
>
> @drake-jenkins-bot linux-gcc-experimental-matlab-ros please
> The results will be available here: https://drake-jenkins.csail.mit.edu/view/Experimental/job/linux-gcc-experimental-matlab-ros/.

jamiesnape, 2016-09-29:
Just above here: http://drake.mit.edu/from_source_ros.html#running-an-example-car-simulation

liangfok, 2016-09-29:
Sorry, I posted the wrong link. I wanted to updated 'jenkins.rst`. PTAL.

david-german-tri, 2017-01-13:
This was done at some point: http://drake.mit.edu/jenkins.html?highlight=jenkins

closed 2017-01-13 by david-german-tri