RobotLocomotion/drake #3641: Build errors with Ubuntu 16.04 and clang 3.7 or 3.8

BetsyMcPhail, 2016-09-30:
The set-up script for Ubuntu 16.04 doesn't specify a version of clang. On my test system, clang 3.8 was installed. The build failed with the following error:

```
13:08:52 [ 59%] Building CXX object systems/plants/constraint/CMakeFiles/drakeRigidBodyConstraint.dir/RigidBodyConstraint.cpp.o
13:08:55 /home/ubuntu/workspace/linux-xenial-clang-experimental/drake/systems/plants/constraint/RigidBodyConstraint.cpp:2299:48: error: default initialization of an object of const type 'const RigidBodyTree::BodyToWrenchMap<Scalar>' (aka 'const std::unordered_map<const RigidBody *, Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, 6, 1, 0, 6, 1>, std::hash<const RigidBody *>, std::equal_to<const RigidBody *>, Eigen::aligned_allocator<std::pair<const RigidBody *const, Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, 6, 1, 0, 6, 1> > > >') without a user-provided default constructor
13:08:55   const RigidBodyTree::BodyToWrenchMap<Scalar> no_external_wrenches;
```

Next, I tried uninstalling clang 3.8 and installing clang 3.7. I received the same error.

Note: I did not get this error with gcc-5.0

rpoyner-tri, 2016-09-30:
On 16.04, i needed to move to clang-3.9 to build drake. Once I did that, I've have no further problems.

BetsyMcPhail, 2016-09-30:
Thanks for the suggestion, I will give it a try.

jwnimmer-tri, 2016-12-22:
I believe this was fixed in #3697.

closed 2016-12-22 by jwnimmer-tri