RobotLocomotion/drake #3646: debugging visualizers for eigen matrices?

RussTedrake, 2016-10-03:
The discussion about matlab vs c++ brought up this idea:  can we find a better way to visualize Eigen matrices in the c++ IDE debuggers?  The ability to quickly inspect the contents of a matrix-valued variable is one major advantage of working in the matlab debugger (changing those values with the interactive shell is another, and is possible but only to an extent in the c++ debuggers).

A quick search turned up this capability for MSVC:
http://eigen.tuxfamily.org/index.php?title=Developer%27s_Corner#Debugging_under_Visual_Studio

Can we achieve something similar for our linux/mac tools?  Selfishly I’d like one for CLion.

Found a page where someone asked about custom debug visualizers for CLion, and the response was “Because CLion uses GDB as a debugger now, you can provide a custom GDB extension and set it in gdbinit.”.  CLion uses LLDB by default, I think, but can use GDB also.  @tkoolen gave mex debugging instructions for LLDB in the CLion docs.  I’d be inclined to stick with it.

It looks like this is possible in LLDB as well.  http://lldb.llvm.org/varformats.html

RussTedrake, 2016-10-03:
ooh.  good news!

just found this for LLDB:
https://github.com/tehrengruber/LLDB-Eigen-Data-Formatter

and this for GDB:
http://dlgreenwald.weebly.com/blog/first-post
http://stackoverflow.com/questions/25085465/using-gdb-with-eigen-c-library

tkoolen, 2016-10-03:
Thanks for looking into this, I never figured this out and it will be very useful.

Related CLion issue: https://youtrack.jetbrains.com/issue/CPP-1798 (please vote for CLion issues if you want them to prioritize something). Probably not 100% necessary given the installation instructions above, but would still be very nice to have for other, non-Eigen types. I believe they did just add (predefined, not custom) pretty printing of STL containers.

CLion uses lldb by default on OSX, and gdb by default on linux. You have the option to use either debugger on either platform, but lldb on linux is currently experimental I believe.

Mathworks actually broke the mex file debugging as described in those instructions in ~2014b. I opened an issue with them, but they didn't take any action to fix it. We should probably just get rid of those instructions; I'll open an issue.

edrumwri, 2016-10-12:
Funny- we were independently looking for a solution in CLion for this problem this morning. I think the simplest answer is a CLion plugin for Eigen inspection. I could invest some time in this if folks have not already moved on.

tkoolen, 2016-10-12:
That would be great if it's not too much work!

RussTedrake, 2016-10-12:
i'm on mac, so was definitely going to try the LLDB viewer, but have not done it yet.  by all means, beat me to it.

edrumwri, 2016-10-13:
Ok, will work on it this week.

RussTedrake, 2017-08-01:
@SeanCurtis-TRI -- didn't you contribute some know-how, and perhaps even code, along these lines?  (but i couldn't find it on slack nor github on a quick search)

SeanCurtis-TRI, 2017-08-01:
I did indeed. You can give this a try:

https://github.com/SeanCurtis-TRI/drake_gdb

If you encounter problems or feel there are missing features, let me know.

fantaosha, 2018-02-06:
I just wrote a LLDB pretty printer for Eigen 

https://github.com/fantaosha/LLDB-Eigen-Pretty-Printer

It has all features as the GDB-Eigen-Pretty-Printer https://github.com/RLovelett/eigen/tree/master/debug/gdb

RussTedrake, 2018-11-01:
I think the solutions posted here are as much as we'll have for now.  Thanks! Closing this one.

closed 2018-11-01 by RussTedrake

dmillard, 2019-07-23:
Sorry for posting on a closed issue, but it comes up when searching for "eigen gdb"

https://github.com/dmillard/eigengdb

It formats using `numpy`'s formatting code, which many people seem to tolerate:

```
(gdb) p mat
$1 = Eigen::Matrix<double,4,4,ColMajor> (data ptr: 0x7fffffffdde0)
[[ 0.68037543  0.82329472 -0.44445058 -0.27043105]
 [-0.21123415 -0.60489726  0.10793991  0.02680182]
 [ 0.56619845 -0.32955449 -0.0452059   0.90445945]
 [ 0.59688007  0.53645919  0.25774185  0.83239014]]
```