RobotLocomotion/drake #3647: plotting tool for c++ development

RussTedrake, 2016-10-03:
Another big feature of working in matlab is the ability to easily plot data.  I have a stub of code that I was working on that, I think, could provide basically the same functionality in c++.  Essentially it is an matlab lcm client which would work very much like bot-lcmgl… you would publish matlab plotting commands via lcm and have a matlab window open that was listening for them and running them from the console.

another known option is using gnuplot (e.g. via some c wrapper http://ndevilla.free.fr/gnuplot/gnuplot_i/index.html ).  do people have other visualizer tools in c++ already?

ggould-tri, 2016-10-03:
For Schunk development I built a generic LCM logger+plotter in python using matplotlib.  It is primitive but it exists.  See `lcm_logger` and `lcm_plot` in https://github.com/RobotLocomotion/drake-schunk-driver/tree/master/tools .  Note that that's really code that I wrote for my own use and is not very polished.

siyuanfeng-tri, 2016-10-03:
A step back, I really wanted something that "flattens" the lcm log. I gave it a list of "traces" that I want to flatten it gives a dense flat array of numbers. And then we can do whatever plotting thing we want later. 
It should be generic to different lcm types, should be able to handle cascaded lcm messages, interpolates between msgs to fill gaps. 
This will be tremendously helpful for working with robots. I am all fine with visualizing data online, but I see greater value to be able to process data offline in batch mode.

liangfok, 2016-10-10:
For reference, here are the various plotting backends used by ROS's [rqt_plot](http://wiki.ros.org/rqt_plot):

[image]

RussTedrake, 2016-10-10:
following those links, i also found:
http://mathgl.sourceforge.net/doc_en/Main.html (GPL/LGPL)

but i'm leaning towards just using VTK directly.  http://www.vtk.org/Wiki/VTK/Examples/Cxx/Plotting/LinePlot

edrumwri, 2016-10-11:
Any reason why no one is mentioning the obvious replacement for MATLAB plotting, matplotlib (Python)? It's callable from C++, in theory at least.

liangfok, 2016-10-11:
Yeah, ROS's `rqt_plot` has "MatPlot" as one of its options, which is based on MatPlotLib. In the GUI snapshot above, it states that MatPlot "needs most CPU". Not sure why, but it's something to consider. Maybe for a quick fix we can use VTK's plotting capabilities to avoid needing to add another dependency, and then longer term create a framework for allowing a number of plotting back ends to be used.

edrumwri, 2016-10-11:
Ah, yes, I see!  My experience is also that it's a little CPU intensive, but that it presents the closest plotting interface to MATLAB's.

liangfok, 2016-10-12:
Should the plotting tool be "loosely coupled" with the simulation in terms of only working with published LCM messages? The alternative would be to "tightly couple" the plotting tool by implementing it as a System 2.0 system that can be wired into a Diagram and directly fed the `Context` containing the information to be plotted.

RussTedrake, 2016-10-12:
We already have the "loosely coupled" version that works on LCM types:  https://github.com/openhumanoids/signal-scope .  lcm-spy also has recently improved its plotting functionality.

liangfok, 2016-10-12:
Aha! Now I better understand the need for https://github.com/RobotLocomotion/drake/issues/3734. I assume, then, that a `Trajectory` object will be provided to the plotting tool, which then displays the trajectory contained within that object?

RussTedrake, 2016-10-12:
you've got it.

liangfok, 2016-10-24:
Related project based on Qt5: http://discourse.ros.org/t/early-preview-plotjuggler-rqt-plot-and-rqt-bag-on-steroids/714/1

patmarion, 2016-11-11:
The application in the above link, PlotJuggler, looks quite full featured.  It seems you could build that without the ROS support and add an LCM source to it.

RussTedrake, 2017-04-25:
say a nice talk about a tool for simple interactive visualization of static data: https://vega.github.io/vega-lite/

RussTedrake, 2018-11-06:
I think that the python bindings have satisfactorily resolved this.

closed 2018-11-06 by RussTedrake