RobotLocomotion/drake #3648: win32 compilation fails on functional_form.h

RussTedrake, 2016-10-03:
```
(ClCompile target) ->
  Z:\russt\drake-distro-win32\drake\..\drake/common/functional_form.h(685): fatal error C1001: An internal error has occurred in the compiler. [Z:\russt\drake-distro-win32\build\drake\common\test\functional_form_test.vcxproj]
```
My CMakeCache is attached.  
[CMakeCache.txt](https://github.com/RobotLocomotion/drake/files/506060/CMakeCache.txt)

RussTedrake, 2016-10-03:
possibly related
```
  functional_form.cc
Z:\russt\drake-distro-win32\drake/common/functional_form.h(546): warning C4624: 'drake::FunctionalForm::Variable::<unnamed-tag>': destructor was implicitly defined as deleted because a base class destructor is inaccessible or deleted [Z:\russt\drake-distro-win32\build\drake\common\drakeCommon.vcxproj]
 ```

bradking, 2016-10-03:
@RussTedrake what git sha1 do you have checked out?  Which build configuration (Debug, Release, etc.) are you building in VS?

RussTedrake, 2016-10-03:
it happened with the master SHA.  Both Release and Debug modes.  Building from cmake —build at the command line (via VS).

bradking, 2016-10-03:
I've built commit `139d6077` as both `Debug` and `Release` with VS 2015 (x86 arch) and cannot reproduce the ICE.

The next step is to try to compile your exact translation unit.  Please open the project in the VS IDE, expand the `functional_form_test` project, and find the `functional_form_test.cc` source.  Right-click and navigate `Properties -> C/C++ -> Preprocessor -> Preprocess to a File` and choose `Yes (/P)`.  Apply and quit the dialog.  Then right-click again and choose `Compile`.  Finally, navigate again and turn the option back off.  Then in the build tree you should be able to find a `.i` file with the preprocessor output for the translation unit that ICEs.  Please attach that here.

RussTedrake, 2016-10-04:
@bradking — thank you.  but if it’s turning out to be this difficult then I think we should drop this, since we are dropping support for msvc anyhow.  It could be some quirk of my virtual machine (running out of memory handling the templates, etc).

bradking, 2016-10-05:
> I think we should drop this, since we are dropping support for msvc anyhow.

Okay, closing for now then.  We can re-open if needed in the future.

closed 2016-10-05 by bradking