RobotLocomotion/drake #3658: Wrap vector with the frame

hongkai-dai, 2016-10-03:
This subject is discussed for several times with @tkoolen , @siyuanfeng-tri , @mitiguy , @sherm1 . It is desirable to say in which frame the vector is expressed in, since the same physical vector can have different numerical values when the frames are different. Also when using a transformation, like `Eigen::Isometry3d`, we should add the information on from which frame the vector is transformed, and to which frame the vector is transformed. The best way I can think of to handle this complexity is to create a class, that stores both the vector and the frame. Also this class should know how to transform a vector from one frame to another frame.

A mini-implementation of this class may look like
```
template<int Dim, typename T>
Class VectorWithBases {
private:
  Matrix<T, Dim, 1> vec_;
  Matrix<T, Dim, Dim> bases_; // bases_.col(0) is the first basis vector, bases_.col(1) is the second basis vector, etc.
public:
  VectorWithBases(const Matrix<T, Dim, 1>& vec, const Matrix<T, Dim, Dim>& bases);
  ExpressedIn(const Matrix<T, Dim, Dim>& new_bases);
}
```

This should work for 2D and 3D vectors. 

For 6D Plucker vector, we should have a different class
```
template<typename T>
Class PluckerVectorWithBases {
private:
  Matrix<T, 6, 1> vec_;
  Matrix<T, 3, 3> bases_; // bases_.col(0) is the first basis vector, bases_.col(1) is the second basis vector
  Matrix<T, 3, 1> origin_; // origin_ is the origin of the Plucker coordinates.
public:
  PluckerVectorWithBases(const Matrix<T, 6, 1>& vec, const Matrix<T, 3, 3>& bases, const Matrix<T, 3, 1> &origin_);
  ExpressedIn(const Matrix<T, 3, 3> &new_bases, const Matrix<T, 3, 1> &origin_);
```

mitiguy, 2016-10-03:
I really like HongKai's suggestion and think it would greatly help the team.
The accuracy and time to delivery (development time) of the multibody codes Interactive Physics, Working Model, visualNastran 4D, were significantly speeded up by a class called
**BVect3** (and related classes BVect2 for 2D analysis, BVect33 for dyadics/tensors, etc).
For example, BVect3 had 3 scalars and a reference to a class called RigidBasis.
The **RigidBasis** class contained information and rotation matrices relating it to other
RigidBasis objects (e.g., the World RigidBasis object).
I would be happy to upgrade/release these classes to TRI and/or public domain
to work within the drake framework (these classes were originally licensed by
me to Knowledge Revolution/MSC.Software - and later obtained by other companies).

sherm1, 2016-10-03:
I think the first step is to add some unambiguous documentation for the existing methods, using unambiguous notation. I think that would avoid a lot of problems. I would like to see how far that gets us before embarking on a more complex set of classes (with unknown performance implications).

tkoolen, 2016-10-03:
See my Julia package for what I would recommend:
* [`Transform3D`](https://github.com/tkoolen/RigidBodyDynamics.jl/blob/master/src/frames.jl#L72)
* [`Twist`](https://github.com/tkoolen/RigidBodyDynamics.jl/blob/master/src/spatial.jl#L130)
* [`GeometricJacobian`](https://github.com/tkoolen/RigidBodyDynamics.jl/blob/master/src/spatial.jl#L182)
* [`SpatialAcceleration`](https://github.com/tkoolen/RigidBodyDynamics.jl/blob/master/src/spatial.jl#L382)
* [`FreeVector3D`, `Point3D`](https://github.com/tkoolen/RigidBodyDynamics.jl/blob/master/src/frames.jl#L32)
* [`Wrench`](https://github.com/tkoolen/RigidBodyDynamics.jl/blob/master/src/spatial.jl#L248)
* [`Momentum`](https://github.com/tkoolen/RigidBodyDynamics.jl/blob/master/src/spatial.jl#L264)
* [`MomentumMatrix`](https://github.com/tkoolen/RigidBodyDynamics.jl/blob/master/src/spatial.jl#L312)
* [`SpatialInertia`](https://github.com/tkoolen/RigidBodyDynamics.jl/blob/master/src/spatial.jl#L1)

tkoolen, 2016-10-03:
One more: [`CartesianFrame3D`](https://github.com/tkoolen/RigidBodyDynamics.jl/blob/master/src/frames.jl#L1).

sherm1, 2016-10-03:
@tkoolen: those variable names like v̇ are awesome! I'm jealous.

tkoolen, 2016-10-06:
FYI: all quantities in `KinematicsCache` have up to this point been expressed in / differentiated in world frame, i.e. the frame associated with the `RigidBodyTree::world()`, by convention. If we're going to use different frames, this issue should receive a higher priority. @SeanCurtis-TRI, until this issue has been resolved, make sure you document the frame in which the contact wrenches are expressed clearly.

hongkai-dai, 2016-11-28:
I would love to work on this issue, but I think I cannot start until the end of this year, hope nobody is really blocked on this now.

hongkai-dai, 2019-03-04:
I don't think we are going to do this.

closed 2019-03-04 by hongkai-dai