RobotLocomotion/drake #3662: A few drake tests fail under valgrind

david-german-tri, 2016-10-03:
See #3656.

- [ ] 46 - valgrind_mathematical_program_test (Failed)
- [ ] 109 - valgrind_model_test (Failed)
- [ ] 138 - valgrind_load_model_test (Failed)
- [ ] 150 - valgrind_rigid_body_tree_test (Failed)

#46 appears to be a bug in nlopt, #109 appears to be a bug in Bullet (previously identified by @amcastro-tri, see https://github.com/RobotLocomotion/drake/blob/9828a841d04ae7bb60d719b78e6e116f7bfc8967/drake/systems/plants/collision/bullet_model.cc#L182), #138 appears to be a bug in gtest itself, and #150 is likely a bug in Drake (but not conclusively proven).

amcastro-tri, 2016-10-04:
cc'ing @SeanCurtis-TRI here

david-german-tri, 2016-12-14:
It is possible that #2710 will address (109) above.

david-german-tri, 2016-12-14:
(46) above was fixed in https://github.com/stevengj/nlopt/pull/89 and #3909

david-german-tri, 2016-12-14:
(150) above was fixed in #3664

david-german-tri, 2017-04-03:
The exact list of failures in this issue is probably obsolete, but we do still have some failures under memcheck.

m-chaturvedi, 2017-04-05:
Here's the current state :

For the Xenial valgrind build the remaining error [ moby_lcp_solver ] seems like a problem with eigen3.  I filed the following bug:
http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1413

For Ubuntu 14.04, the problem seems to be with tests with EXPECT_DEATH.  I've suppressed the errors except the moby_lcp_solver, which is common between Xenial and Trusty.   Relevant link:
https://github.com/RobotLocomotion/drake/pull/5608

For OSX the suppressions from Valgrind source code are not being picked by it.  I've submitted a PR for it [ but there are ~3000 errors still left ]
https://github.com/RobotLocomotion/drake/pull/5682

For non open-source builds, mosek seems to be causing a problem, and here is the PR for it. [ I'm testing if everything goes away ]:
https://github.com/RobotLocomotion/drake/pull/5722

jamiesnape, 2017-04-05:
We also have some test sizes and/or timeouts that will need tweaking, and a subset of the suppressions will need to be applied to the Bazel builds.

jamiesnape, 2017-04-11:
@david-german-tri Given #5785, what do you want to do about the Eigen bug and its potential fix (https://bitbucket.org/eigen/eigen/pull-requests/308/).

stonier, 2017-06-01:
@m-chaturvedi can you rebuild the checklist for what is currently failing?

m-chaturvedi, 2017-06-05:
The following observation are for Bazel everything runs with valgrind [1] [2].
- `rotation_constraint_test` : Fails for gcc but not clang.  Plausibly the same reason why we had to revert: `ec00b6fc6`. 
- `lcm_driven_loop_test`: Fails even without valgrind, has an issue: https://github.com/RobotLocomotion/drake/issues/5936

There are a few other timeouts, not failures though.  The timeouts are only for Clang, no timeout happens for gcc.

Clang timeouts: 
- trajectory_optimization_test 
- piecewise_generation_test
- system_identification_test
- implicit_euler_integrator_test
- runge_kutta3_integrator_test
- semi_explicit_euler_integrator_test

[1] https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-bazel-nightly-memcheck-valgrind-everything/25/
[2] https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-bazel-nightly-memcheck-valgrind-everything/26/

jamiesnape, 2017-08-10:
All tests now pass.

closed 2017-08-10 by jamiesnape