RobotLocomotion/drake #3663: Add "minimal" to pre-merge testing

jwnimmer-tri, 2016-10-03:
We should get "minimal" into the pre-merge tests.  It breaks a lot (relatively speaking), but is _super cheap_ to run.

jwnimmer-tri, 2016-10-03:
We could consider minimal + cpplint as the front-of-pipeline build if that makes things easier, but highest priority is to get `minimal` covered somehow, lower priority to optimize build time and feedback cycle.

david-german-tri, 2016-10-03:
Agreed.

jwnimmer-tri, 2016-10-04:
Breaks almost every day.  Bumping up priority.

jamiesnape, 2016-10-05:
Debug or Release, Clang or GCC?

david-german-tri, 2016-10-05:
Maybe gcc/debug, since we already have clang/release, clang/debug, and gcc/release in other configs?

jamiesnape, 2016-10-05:
My thoughts exactly.

jwnimmer-tri, 2016-10-05:
Whatever you all decide is fine by me.

jamiesnape, 2016-10-05:
Now running concurrently with cpplint on pull requests.

https://drake-jenkins.csail.mit.edu/job/experimental/compiler=gcc,label=linux_minimal,type=debug/

closed 2016-10-05 by jamiesnape

david-german-tri, 2016-10-05:
Having lived with it for the past hour, I'm not actually sure running this concurrent with cpplint was a good idea.  The forced serialization is a noticeable bump in pre-merge CI latency.

jamiesnape, 2016-10-05:
I think I agree.