RobotLocomotion/drake #3665: disable windows CI

RussTedrake, 2016-10-04:
proposal from yesterday’s meeting was that we disable windows CI immediately, without blocking on bringing up the valgrind builds and linux 32-bit builds.  Expectation is that we can quickly fix any bugs introduced in the interim.

Given my pains in #3501 , I’d love to see that happen asap.

jwnimmer-tri, 2016-10-04:
Yes please, ASAP.

liangfok, 2016-10-04:
FTR, #3643 was delayed about 24 hours due to a mysterious Win32 linker error. Fixing it required adding something to the CMakeLists.txt: https://github.com/RobotLocomotion/drake/pull/3643/commits/6def3466e74da3bf617feb4a7ebefc2c31bcd37f#diff-7a7595a8eccbcad1fb24437727b22ebcR34.

david-german-tri, 2016-10-04:
SGTM, I'll go figure out what buttons to click in Jenkins.  I'm thinking - turn it off in presubmit and continuous for the moment, but leave the nightly running (not buildcop-enforced, just keeping track of what if anything breaks) until the valgrind and linux32 builds are up.

david-german-tri, 2016-10-04:
@BetsyMcPhail @jamiesnape FYI - let's coordinate in Slack on #build

RussTedrake, 2016-10-04:
but we would need to be clear that the policy is NOT to pull a build cop revert if windows breaks.

david-german-tri, 2016-10-04:
Yep, exactly.

jwnimmer-tri, 2016-10-04:
Actually, I'd _really_ like to remove the `drakeFoo_export.h` ceremony.  It's actively impeding my work.  So I'd appreciate having nightly disabled as well.

sammy-tri, 2016-10-04:
I assume that would also involve disabling hidden symbols on ELF builds?  (I'm curious what hole you've fallen into, but this issue is not the place for that discussion)

mwoehlke-kitware, 2016-10-04:
> Actually, I'd *really* like to remove the `drakeFoo_export.h` ceremony. It's actively impeding my work.

Export decoration is TRTTD, even on ELF platforms. That said, if necessary we ought to be able to pass the symbol definitions as target usage requirements, which would eliminate the need for a header, but I am opposed to dropping export decorations entirely.

david-german-tri, 2016-10-04:
-> #3666 with the export decorations talk, please; I want to close this one as soon as I'm done throwing the switch.

david-german-tri, 2016-10-04:
And it's done!

closed 2016-10-04 by david-german-tri