RobotLocomotion/drake #3673: SDF-based robots invisibly, unpredictably omit joint outputs

ggould-tri, 2016-10-04:
During SDF parsing, some joints are converted to loop joints, which do not provide any outputs.  The information about which joints have been converted into loop joints is discarded in `parser_sdf.cc`.  Thus user code with an SDF-based model cannot know which joint coordinates are present or absent from the output vector of the RigidBodySystem, and so cannot interpret that output vector at all.

The workaround is to add debug printing to `parser_sdf`'s loop creation and use that to compute offsets into the output vector.

Correct fixes might include:
 * the parser informing its client of which joints got stepped on,
 * generation of outputs from loop joints, or
 * an sdf "plugin" to tell the parser which joints to loop on.  (This solution is bad!  Don't do it!)

Because RigidBodySystem is in the midst of a rewrite, this is probably a low-priority issue.  However the implementers of the replacement MultibodyWorld or whatever it is eventually called should take care not to repeat this error.

Mentioning @sherm1 and @amcastro-tri as likely interested in this issue.

ggould-tri, 2016-10-04:
Actually I think it's a bit worse than that:  AIUI, joints with no inertia leafward of them in the tree are also omitted from the state, so different sets of joints will be omitted from the state depending on where the loop cut lands.  This means that in some cases the output vector's length will depend on where the loop is cut, which of course is not documented, returned, or particularly predictable.

sherm1, 2016-10-04:
See issue #1867 for more information about handling loop joints sensibly.

ggould-tri, 2016-10-05:
#1867 is definitely yet another correct and sensible way to solve this problem!  As long as one of them happens, I don't care much which one.

(as a sidenote, #1867 doesn't do anything to address the general difficulty of correlating components of an sdf to the members of input/output vectors of the system; that will definitely need to be addressed sooner or later, but maybe that is already in the mix for the multibody world design.)

sherm1, 2016-10-05:
For correlating sdf joints with their coordinates we should be able to (1) obtain a joint object from its sdf name, then (2) ask the joint for its coordinates. #1867 can provide part of that by defining the mapping between the named sdf joints and the multibody tree joints chosen to represent them. Decorating the output ports conveniently with that info remains to be resolved; there was lots of related discussion in the RigidBodyPlant outputs PR #3471.

SeanCurtis-TRI, 2016-11-30:
@ggould-tri I have a question for you on this.  

I'm mucking around in `parser_sdf.cc` and saw this [line](https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/parser_sdf.cc#L610) of code and the corresponding note pointing to this issue.

It appears the logging statement is in the wrong place; it is set to print out if the child body of a joint has a collision element.  If the body has no collision element, there will be no print statement for the joint being referenced.  Is that causing you problems?  The line should be moved to [here](https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/parser_sdf.cc#L677).

If you're not in a hurry, I'll include that move in my own PR.  Otherwise you can feel free to do it.

Also, @sherm1 recommended changing the text to refer to it as a "tree joint" rather than just "joint".  Would changing the wording of the message mess you up?

amcastro-tri, 2016-12-01:
I agree with @sherm1 solution here. But also, should the user know about the generalized coordinates and their mappings at all? My plan is to provide an API that looks like (or similar variations):

```c++
RigidBodyPlant plant("my_model.sdf"); // This is pseudo-code, in reality we'd use a parser.
Context<T> context = plant.CreateDefaultContext();
const auto& elbow_joint = plant.GetJointByName("elbow");
T elbow_joint_angle = elbow_joint.get_angle(context);
// And similar API's for RigidBody's
const auto& forearm = plant.GetBodyByName("forearm");
Vector3<T> xc = forearm.get_center_of_mass_world(context);
```

amcastro-tri, 2016-12-01:
I created #4328 to track the development of our multibody engine's API to users.

liangfok, 2017-03-31:
This issue is currently blocked on #4337 (switch SDF parsers) and #4633 (MBT).

@amcastro-tri, what do you think about closing this issue and adding it as a wish-list item to #5013?

amcastro-tri, 2017-03-31:
Added item to #5013. Thanks

closed 2017-03-31 by amcastro-tri