RobotLocomotion/drake #3675: Add functions to return MathematicalProgram object inside inverseKin and inverseKinTraj

hongkai-dai, 2016-10-04:
Currently `inverseKin` and `inverseKinTraj` construct a `MathematicalProgram` object inside the functions, and then solve the nonlinear optimization within the function calls. It can be beneficiary to create a function `constructsInverseKinMathematicalProgram` in `RigidBodyIK.h`, so that the user can access and modify the underlying nonlinear optimization problem.

The reason why we want to expose the nonlinear optimization to the user, is that the user may want to add more cost, constraints, or decision variables to the nonlinear optimization problem, which are not supported by `inverseKin` or `inverseKinTraj`. One example is that if we want the robot center of mass to be close to the center of the foot region, then we want to add a cost, to penalize the largest weight in the `QuasiStaticConstraint`. This cost would look like
```
min max_weight
// The first four constraints are the current QuasiStaticConstraint
com_x = sum_i foot_vertex_x_i * weight_i
com_y = sum_i foot_vertex_y_i * weight_i
weight_i >= 0
sum_i weight_i = 1
// add max_weight as the new decision variable
max_weight >= weight_i 
```

So if we create `ConstructsInverseKinMathematicalProgram` and `ConstructsInverseKinTrajMathematicalProgram`, the user can solve the IK problem in either of the two approaches

1. Call `inverseKin(plant, q_nominal, q_guess, constraint)` directly, as the current API, where inside `inverseKin`, it calls `ConstructsInverseKinMathematicalProgram`, and then solve this nonlinear optimization problem.
2. The user can also take the following steps

   i. Constructs a `MathematicalProgram` object as `prog = ConstructsInverseKinMathematicalProgram(plant, q_nominal, constraint)`.
  ii. Do something as the user wish, like add more constraints, costs or decision variables.
  iii. Solve the IK problem by `solution = prog.solve()`.

patmarion, 2016-10-10:
Yes, this would be great and I would love to have this.

I think the approach outlines in 2. sounds good, but also what about an interface that is similar to 1., but in addition to the contraints the user passes a list of cost objectives?  Then some of the commonly used objectives could be provided by drake and not left up to the user code.  Just like we have a WorldPositionConstraint, there could be a WorldPositionCost, JointPositionCost, etc.  This way the user can still program in terms of inverse kinematics, instead of in terms of MathematicalProgram.

hongkai-dai, 2016-10-10:
That is a good idea. I like it! Thanks @patmarion 

A side note, WorldPositionConstraint might not be a good example. The `Eval` function of WorldPositionConstraint just computes the position of the end effector, in the world frame. I think you want some constraint that returns the **deviation** from the desired pose, like WorldGazeDirConstraint, which computes the angle between the actual gaze direction and the desired gaze direction.

hongkai-dai, 2019-01-25:
inverseKin and inverseKinTraj are outdated.

closed 2019-01-25 by hongkai-dai