RobotLocomotion/drake #3683: Race Condition Between Generated C Language LCM Message Dependencies

liangfok, 2016-10-05:
# Problem Definition

There are often dependencies between Drake's LCM messages. For example, [lcmt_qp_controller_input](https://github.com/RobotLocomotion/drake/blob/master/drake/lcmtypes/lcmt_qp_controller_input.lcm) depends on [lcmt_zmp_data](https://github.com/RobotLocomotion/drake/blob/master/drake/lcmtypes/lcmt_zmp_data.lcm). If the dependencies are not correctly specified, race conditions can occur where an LCM message is compiled before all of its dependent LCM messages are generated. An example race condition is here: 

https://drake-jenkins.csail.mit.edu/job/linux-gcc-continuous-matlab-open-source/782/console

The relevant part of the log file is copied below:

```
10:34:20 /home/ubuntu/workspace/linux-gcc-continuous-matlab-open-source/build/drake/lcmtypes/drake_lcmt_qp_controller_input.h:18:33: fatal error: drake_lcmt_zmp_data.h: No such file or directory
10:34:20  #include "drake_lcmt_zmp_data.h"
10:34:20                                  ^
10:34:20 compilation terminated.
10:34:21 make[2]: Entering directory `/home/ubuntu/workspace/linux-gcc-continuous-matlab-open-source/build/drake'
10:34:21 [  7%] Generating drake_lcmt_zmp_data.h, drake_lcmt_zmp_data.c, drake/lcmt_zmp_data.hpp, drake/lcmt_zmp_data.py, drake/lcmt_zmp_data.java
```

The above console output snippet indicates that the compiler tried to find `drake_lcmt_zmp_data.h` before it was generated. Looking in [drake-distro/drake/lcmtypes/CMakeLists.txt](https://github.com/RobotLocomotion/drake/blob/master/drake/lcmtypes/CMakeLists.txt), there is apparently code that ensures the C++ code is generated before building compiled:

https://github.com/RobotLocomotion/drake/blob/master/drake/lcmtypes/CMakeLists.txt#L49

However, no such logic appears to exist for the generated C code. Maybe similar logic must be provided for the C code?

liangfok, 2016-10-05:
@mwoehlke-kitware I'm going to assign this to you since I believe you are most familiar with how this C library is built, based on this commit: 6fcf875cc91308fa3b26549571c6c0c2eef43fe2

mwoehlke-kitware, 2016-10-05:
After talking to @bradking, this should only affect ninja (or at least, not make); dependency scanning is *supposed* to account for it...

liangfok, 2016-10-05:
I see. Can we add dependency scanning to `ninja`? I suppose disabling `ninja` is a non-starter?

PR #3962 (2016-11-01): LCM types build updates
Remove superfluous dependencies on `drake_lcmtypes_hpp` (which no longer exists). Update how we build our LCM types library to use all the shiny new stuff from LCM upstream.

Fixes #3683.

@liangfok for feature review.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3962)
commit 8094948f (2016-11-01): Refactor build of LCM types

Update how we build our LCM types library to use all the shiny new stuff
from LCM upstream. This gives us proper exports, which allows us to
build the library shared (if enabled) rather than forcing it to be built
static, fixes a possible race condition in the build, and generally
makes the logic more streamlined.

Fixes #3683.
closed 2016-11-01 by mwoehlke-kitware