RobotLocomotion/drake #3684: addpath_drake fails, needs a folder added to path

mposa, 2016-10-05:
In MATLAB, addpath_drake fails with the error message:

> Undefined function or variable 'get_drake_binary_dir'.
> 
> Error in addpath_drake (line 63)
> bindings_dir = fullfile(get_drake_binary_dir(),'bindings','matlab');

From @tkoolen , it seems to require that **drake-distro/build/install/matlab** be added to the path first.

david-german-tri, 2016-10-05:
> From @tkoolen , it seems to require that drake-distro/build/install/matlab be added to the path first.

That is supposed to happen automatically, and if it doesn't you're supposed to get an informative error:

https://github.com/RobotLocomotion/drake/blob/master/drake/addpath_drake.m#L10

So, I think something else has gone wrong that is not apparent from your post.

mposa, 2016-10-05:
I see. That line wasn't being called because a previous version of drake didn't clean up it's path. Thanks.

closed 2016-10-05 by mposa