RobotLocomotion/drake #3701: Intent to replace internal OBJ loader

nkoenig, 2016-10-06:
Drake has a custom OBJ loader in `DrakeShapes::Mesh`. We can replace this loader with [tinyobjloader](https://github.com/syoyo/tinyobjloader). 

The tinyobjloader was recently tested in Gazebo [here](https://bitbucket.org/osrf/gazebo/pull-requests/2454). The plan is to do something similar with Drake.

tinyobjloader would be placed in a subfolder under `externals`, and `DrakeShapes::Mesh` would make use of this loader.

@sherm1 , do you have any thoughts or comments on this plan?

sherm1, 2016-10-06:
Thanks, Nate! Tinyobjloader looks awesome. I believe the right way to include this dependency in Drake would be as a new git submodule (those do end up under `externals`). I would much prefer this to a homebrew obj reader, but I don't actually know our current level of satisfaction with that. @liangfok, @SeanCurtis-TRI, any thoughts on this?

liangfok, 2016-10-06:
:+1: for making `tinyobjloader` a git submodule of `Drake`.

SeanCurtis-TRI, 2016-10-06:
I chatted with @sherm1.  While the principle of us not having our own obj reader, there are a couple of issues.  We *do* already have one.  It's tied into drake.  So, switching for the sake of switching is going to have to address that change in interface.  Is there a pain point this addresses beyond us not writing our own?  It may make sense to defer this until we know have a problem with our own obj reader -- i.e., if any problem comes up with our current state, rather than investing the energy into patching our ad hoc reader, we should invest that energy in having someone else solve our problems.

jwnimmer-tri, 2016-10-06:
My expectation is that `class Mesh` APIs would remain identical, but would now be backed by a well-tested implementation, instead of the barely-good-enough-and-untested #2782 like we have now.

SeanCurtis-TRI, 2016-10-06:
Agreed -- that means integration with tinyobjreader would essentially be implementing a mapping between the library's representation and our own.

And, other than the principle, do we have a practical need to do so?  Or, more particularly, there's no better use of resources at this point?  Obviously, if the answer to either of those is yes, then it's a slam dunk.  But otherwise, it's not so clear to me.

liangfok, 2016-10-06:
I'm currently not blocked on this and am not aware of anyone being blocked on it. I'll thus assign a low priority to this issue.

jwnimmer-tri, 2016-10-06:
> And, other than the principle, do we have a practical need to do so? 

As the automotive side begins simulating additional perception sensors and larger, more interesting worlds, it's likely helpful if our foundations are solid.  Not worrying about `.obj` loading bugs seems helpful in that light.

> Or, more particularly, there's no better use of resources at this point?

Well, in point of fact, we are somewhat constrained in how we can apply OSRF staffing.  This one is a good fit.  They recently added `.obj` support to Gazebo for us, so are already spun up on this library.

jwnimmer-tri, 2016-10-06:
(To be clear, it was at my instruction that @nkoenig is looking into this; I asked him to file this issue to raise awareness / facilitate discussion.)

SeanCurtis-TRI, 2016-10-06:
Suits me.  Having expressed some thoughts, I happily acquiesce to whatever the need is.

jwnimmer-tri, 2016-10-06:
Great.

Good that the "don't change the APIs (in the first PR)" got articulated for whoever grabs this issue.

sherm1, 2016-10-06:
I like @jwnimmer-tri's arguments for plowing ahead!

stonier, 2017-03-31:
Added in PR #5631.

closed 2017-03-31 by stonier