RobotLocomotion/drake #3703: CMake Option -DUSE_SYTEM_VTK=OFF Causes Build to Fail

liangfok, 2016-10-06:
Per https://github.com/RobotLocomotion/drake/issues/2624#issuecomment-251937730, I modified `drake-distro/CMakeLists.txt` to not use the system's VTK:

https://github.com/liangfok/drake/commit/d810ee4e780562ee4cdc155c744879e569558f48

Unfortunately, with this modification, Drake's build fails with the following error:

```
[10/246] Generating moc_ddLCMThread.cxx
FAILED: cd /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/src/vtk/DRCFilters && /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/src/vtk/DRCFilters/depth_compose.cxx /home/liang/dev/drake-distro-1/externals/director/src/vtk/DRCFilters/edl_resources/Shaders/depth_compose.glsl depth_compose
/bin/sh: 1: /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/src/vtk/DRCFilters/depth_compose.cxx: not found
FAILED: cd /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/src/vtk/DRCFilters && /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/src/vtk/DRCFilters/bilateral_filter.cxx /home/liang/dev/drake-distro-1/externals/director/src/vtk/DRCFilters/edl_resources/Shaders/bilateral_filter.glsl bilateral_filter
/bin/sh: 1: /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/src/vtk/DRCFilters/bilateral_filter.cxx: not found
FAILED: cd /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/src/vtk/DRCFilters && /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/src/vtk/DRCFilters/edl_compose.cxx /home/liang/dev/drake-distro-1/externals/director/src/vtk/DRCFilters/edl_resources/Shaders/edl_compose.glsl edl_compose
/bin/sh: 1: /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/src/vtk/DRCFilters/edl_compose.cxx: not found
FAILED: cd /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/src/vtk/DRCFilters && /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/src/vtk/DRCFilters/edl_shade.cxx /home/liang/dev/drake-distro-1/externals/director/src/vtk/DRCFilters/edl_resources/Shaders/edl_shade.glsl edl_shade
/bin/sh: 1: /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/src/vtk/DRCFilters/edl_shade.cxx: not found
```

patmarion, 2016-10-06:
Can you post the output of these two grep commands:

    grep USE_SYSTEM_VTK /home/liang/dev/drake-distro-1/build/externals/director/CMakeCache.txt
    
    grep VTK_DIR /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/CMakeCache.txt

And can you confirm that vtk-5.10 was indeed compiled and that this directory exists:

    /home/liang/dev/drake-distro-1/build/install/lib/vtk-5.10

liangfok, 2016-10-06:
Here are the requested outputs:

```
$ grep USE_SYSTEM_VTK /home/liang/dev/drake-distro-1/build/externals/director/CMakeCache.txt
USE_SYSTEM_VTK:BOOL=ON

$ grep VTK_DIR /home/liang/dev/drake-distro-1/build/externals/director/src/director-build/CMakeCache.txt
VTK_DIR:PATH=/usr/lib/cmake/vtk-6.2

$ ls -lah /home/liang/dev/drake-distro-1/build/install/lib/vtk-5.10
ls: cannot access '/home/liang/dev/drake-distro-1/build/install/lib/vtk-5.10': No such file or directory
```

I think you just found the problem! vtk-5.10 isn't being downloaded and compiled.

patmarion, 2016-10-06:
Can you please edit this file:

    /home/liang/dev/drake-distro-1/externals/director/src/vtk/DRCFilters/CMakeLists.txt

And add this cmake code after line 37:

    if(NOT EXISTS "${VTK_ENCODESTRING_EXE}")
      message(SEND_ERROR "The required vtkEncodeString executable was not found.\nThis may indicate an error in your VTK install or configuration.")
    endif()

Then run make in build/externals/director, and let me know if that helps catch the error at configure time (rather than the  unhelpful error at build time that you posted).  If that helps, I'll commit that code, because this is not the first time I have seen this error and it always looks very mysterious.

patmarion, 2016-10-06:
So why is USE_SYSTEM_VTK set to ON even after your modification?  I'll leave that one up to you to debug.  Let me know!

liangfok, 2016-10-06:
Here's the output after modifying `drake-distro/externals/director/src/vtk/DRCFilters/CMakeLists.txt` line 37:

```
CMake Error at src/vtk/DRCFilters/CMakeLists.txt:40 (message):
  The required vtkEncodeString executable was not found.

  This may indicate an error in your VTK install or configuration.
```

Looks like it does catch the error at configure time.

Good point about why `USE_SYSTEM_VTK` is still being set to `ON` even after my modification. I'll look into it. Thanks again.

patmarion, 2016-10-06:
Great.  Thanks for testing this, by the way!  It's probably been quite some time since this mode was actually tested, since most users apt-get install vtk5 or homebrew install vtk5, but at one point in history this mode did work correct, so hopefully it's still good.

liangfok, 2016-10-06:
No problem. It turns out my earlier modification had a typo, which I fixed in https://github.com/liangfok/drake/commit/0a0bd842ed9f666f31b4231aee619bf508d30611. With this fix, the `USE_SYSTEM_VTK` is now being correctly passed into Director's `CMakeCache.txt`:

```
$ grep USE_SYSTEM_VTK /home/liang/dev/drake-distro-1/build/externals/director/CMakeCache.txt
USE_SYSTEM_VTK:BOOL=OFF
```

Unfortunately, the build fails when compiling VTK:

```
In file included from /usr/include/GL/glx.h:333:0,
                 from /home/liang/dev/drake-distro-1/build/externals/director/src/vtk/Rendering/vtkXOpenGLRenderWindow.cxx:31:
/usr/include/GL/glxext.h:480:143: error: ‘GLintptr’ has not been declared
 typedef void ( *PFNGLXCOPYBUFFERSUBDATANVPROC) (Display *dpy, GLXContext readCtx, GLXContext writeCtx, GLenum readTarget, GLenum writeTarget, GLintptr readOffset, GLintptr writeOffset, GLsizeiptr size);
```

Here is the version of VTK that it's trying to compile:

```
$ cd /home/liang/dev/drake-distro-1/build/externals/director/src/vtk
$ git remote -v
origin	https://github.com/bilke/VTK.git (fetch)
origin	https://github.com/bilke/VTK.git (push)
$ git status
HEAD detached at b753b7f
```

Here is the version of OpenGL that was involved in the build error:

```
$ dpkg -S /usr/include/GL/glx.h
mesa-common-dev:amd64: /usr/include/GL/glx.h

$ dpkg -s mesa-common-dev
Package: mesa-common-dev
Status: install ok installed
Priority: optional
Section: libdevel
Installed-Size: 2953
Maintainer: Ubuntu X-SWAT <ubuntu-x@lists.ubuntu.com>
Architecture: amd64
Multi-Arch: same
Source: mesa
Version: 11.2.0-1ubuntu2.2
Depends: libx11-dev, libdrm-dev (>= 2.4.65)
Description: Developer documentation for Mesa
 This package includes the specifications for the Mesa-specific OpenGL
 extensions, the complete set of release notes and the development header
 files common to all Mesa packages.
Homepage: http://mesa3d.org/
Original-Maintainer: Debian X Strike Force <debian-x@lists.debian.org>
```

@patmarion, any ideas what could be wrong? Thanks!

patmarion, 2016-10-06:
Perhaps try this https://github.com/UV-CDAT/uvcdat/issues/983#issuecomment-71814770

Directory is grabbing someone's patched version of vtk 5.10 (which fixes the Windows build), so perhaps we'll need to also make a patch to fix the Ubuntu 16 build.  I can help debug more when I have an Ubuntu 16 machine handy, which I might have next week.

liangfok, 2016-10-06:
Thanks! That fixed work. I was able to build Director on Ubuntu 16.04 with ROS Kinetic installed. I will next try to build Drake.

patmarion, 2016-10-06:
great.  Also you will need to test the python runtime.  Normally director tries to set the python path to guarantee that it imports the correct version of vtk with "import vtk", so hopefully it works, but it might only be tuned for homebrew vtk.  To test director's ability to import vtk correctly, run this command:

    build/install/bin/directorPython -m director.consoleapp

liangfok, 2016-10-07:
I was able to build Drake but ran into the following problem while trying to start `drake-visualizer`:

```
$ cd drake-distro
$ ./build/install/bin/drake-visualizer
./build/install/bin/drake-visualizer: error while loading shared libraries: libvtkPythonCore.so.5.10: cannot open shared object file: No such file or directory
```

The file exists, but the dynamic linker can't find it:

```
$ cd drake-distro
$ find . -name "libvtkPythonCore.so.5.10"
./build/install/lib/vtk-5.10/libvtkPythonCore.so.5.10
./build/externals/director/src/vtk-build/bin/libvtkPythonCore.so.5.10
```

Here are the `libvtkPython*` libraries that the dynamic linker can find:

```
$ sudo /sbin/ldconfig -p | grep libvtkPython
	libvtkPythonInterpreterTCL-6.2.so.6.2 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkPythonInterpreterTCL-6.2.so.6.2
	libvtkPythonInterpreterTCL-6.2.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkPythonInterpreterTCL-6.2.so
	libvtkPythonInterpreterPython27D-6.2.so.6.2 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkPythonInterpreterPython27D-6.2.so.6.2
	libvtkPythonInterpreterPython27D-6.2.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkPythonInterpreterPython27D-6.2.so
	libvtkPythonInterpreter-6.2.so.6.2 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkPythonInterpreter-6.2.so.6.2
	libvtkPythonInterpreter-6.2.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libvtkPythonInterpreter-6.2.so
```

I manually add the directory that contains `libvtkPythonCore.so.5.10`:
```
$ sudo /sbin/ldconfig -v /home/liang/dev/drake-distro-1/build/install/lib/vtk-5.10
$ sudo /sbin/ldconfig -p | grep libvtkPythonCore
	libvtkPythonCore.so.5.10 (libc6,x86-64) => /home/liang/dev/drake-distro-1/build/install/lib/vtk-5.10/libvtkPythonCore.so.5.10
	libvtkPythonCore.so (libc6,x86-64) => /home/liang/dev/drake-distro-1/build/install/lib/vtk-5.10/libvtkPythonCore.so
```

Unfortunately, `drake-visualizer` now segfaults when it starts:

```
$ cd drake-distro
$ ./build/install/bin/drake-visualizer
Segmentation fault (core dumped)
```
Here is the backtrace using `gdb`:

```
$ cd drake-distro
$ gdb -ex run --args ./build/install/bin/drake-visualizer
...
(gdb) bt
#0  0x00007fff70212225 in QMetaObject::className() const () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007fff70b5873f in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#2  0x00007fff70af2ed7 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#3  0x00007ffff7de74ea in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffde18, env=env@entry=0xc217a0) at dl-init.c:72
#4  0x00007ffff7de75fb in call_init (env=0xc217a0, argv=0x7fffffffde18, argc=1, l=<optimized out>) at dl-init.c:30
#5  _dl_init (main_map=main_map@entry=0x13ea050, argc=1, argv=0x7fffffffde18, env=0xc217a0) at dl-init.c:120
#6  0x00007ffff7dec712 in dl_open_worker (a=a@entry=0x7fffffffb3a0) at dl-open.c:575
#7  0x00007ffff7de7394 in _dl_catch_error (objname=objname@entry=0x7fffffffb390, errstring=errstring@entry=0x7fffffffb398, mallocedp=mallocedp@entry=0x7fffffffb38f, 
    operate=operate@entry=0x7ffff7dec300 <dl_open_worker>, args=args@entry=0x7fffffffb3a0) at dl-error.c:187
#8  0x00007ffff7debbd9 in _dl_open (file=0x13f5680 "/usr/lib/python2.7/dist-packages/vtk/vtkRenderingQtPython.x86_64-linux-gnu.so", mode=-2147483390, 
    caller_dlopen=0x7ffff120988d <_PyImport_GetDynLoadFunc+285>, nsid=-2, argc=<optimized out>, argv=<optimized out>, env=0xc217a0) at dl-open.c:660
#9  0x00007fffeeedaf09 in dlopen_doit (a=a@entry=0x7fffffffb5d0) at dlopen.c:66
#10 0x00007ffff7de7394 in _dl_catch_error (objname=0x65c350, errstring=0x65c358, mallocedp=0x65c348, operate=0x7fffeeedaeb0 <dlopen_doit>, args=0x7fffffffb5d0) at dl-error.c:187
#11 0x00007fffeeedb571 in _dlerror_run (operate=operate@entry=0x7fffeeedaeb0 <dlopen_doit>, args=args@entry=0x7fffffffb5d0) at dlerror.c:163
#12 0x00007fffeeedafa1 in __dlopen (file=<optimized out>, mode=<optimized out>) at dlopen.c:87
#13 0x00007ffff120988d in _PyImport_GetDynLoadFunc () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#14 0x00007ffff12784be in _PyImport_LoadDynamicModule () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#15 0x00007ffff1279300 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#16 0x00007ffff127a55a in PyImport_ImportModuleLevel () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#17 0x00007ffff11f1698 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#18 0x00007ffff123d1e3 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#19 0x00007ffff1313447 in PyEval_CallObjectWithKeywords () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#20 0x00007ffff11e65c6 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#21 0x00007ffff131401c in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#22 0x00007ffff11e2b89 in PyEval_EvalCode () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#23 0x00007ffff12771b4 in PyImport_ExecCodeModuleEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#24 0x00007ffff1277b8f in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#25 0x00007ffff1279300 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#26 0x00007ffff127a55a in PyImport_ImportModuleLevel () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#27 0x00007ffff11f1698 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
[... cut ...]
#30 0x00007ffff11e65c6 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#31 0x00007ffff131401c in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#32 0x00007ffff11e2b89 in PyEval_EvalCode () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#33 0x00007ffff12771b4 in PyImport_ExecCodeModuleEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#34 0x00007ffff1277b8f in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#35 0x00007ffff1278b32 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
```

patmarion, 2016-10-07:
It's importing the wrong vtk python module.  It's getting your system vtk python (version 7), which has conflicting library dependencies and crashes.  See line 8 in your gdb output.

It needs to get the vtk python module (version 5.10) that was compiled and locally installed during the superbuild.  It should just be a matter of setting the PYTHONPATH environment variable to resolve the issue.  Set it to PYTHONPATH=<dir containing local vtk module>:$PYTHONPATH  That is, the dir containing vtk/__init__.py, which should be somewhere in build/install.

But please also see my previous comment.  Director is supposed to do this automatically for you, but it's only been tuned to work on homebrew, not for ubuntu where we've traditionally always preferred system vtk.

liangfok, 2016-10-07:
Oops! I didn't see your post about `directorPython` until now. Here are the possible locations of the VTK Python module 5.10 on my system:

```
$ cd drake-distro
$ $ find . -name "__init__.py" | grep vtk
./build/externals/director/src/vtk/Utilities/vtkTclTest2Py/__init__.py
./build/externals/director/src/vtk/Wrapping/Python/vtk/__init__.py
./build/externals/director/src/vtk/Wrapping/Python/vtk/test/__init__.py
./build/externals/director/src/vtk/Wrapping/Python/vtk/qt4/__init__.py
./build/externals/director/src/vtk/Wrapping/Python/vtk/tk/__init__.py
./build/externals/director/src/vtk/Wrapping/Python/vtk/util/__init__.py
./build/externals/director/src/vtk/Wrapping/Python/vtk/qt/__init__.py
./build/externals/director/src/vtk/Wrapping/Python/vtk/wx/__init__.py
./build/externals/director/src/vtk/Wrapping/Python/vtk/gtk/__init__.py
./build/externals/director/src/vtk-build/Wrapping/Python/vtk/__init__.py
./build/externals/director/src/vtk-build/Wrapping/Python/vtk/test/__init__.py
./build/externals/director/src/vtk-build/Wrapping/Python/vtk/qt4/__init__.py
./build/externals/director/src/vtk-build/Wrapping/Python/vtk/tk/__init__.py
./build/externals/director/src/vtk-build/Wrapping/Python/vtk/util/__init__.py
./build/externals/director/src/vtk-build/Wrapping/Python/vtk/qt/__init__.py
./build/externals/director/src/vtk-build/Wrapping/Python/vtk/wx/__init__.py
./build/externals/director/src/vtk-build/Wrapping/Python/vtk/gtk/__init__.py
```
I modified my `PYTHONPATH` environment variable to point to the version in `vtk-build`:

```
$ cd drake-distro
$ cd build/externals/director/src/vtk-build/Wrapping/Python/
$ export PYTHONPATH=`pwd`:$PYTHONPATH
```

Now when I try to run `directorPython`, it's failing to find `vtkCommonPython`:

```
$ cd drake-distro
$ ./build/install/bin/directorPython -m director.consoleapp
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/liang/dev/drake-distro-1/build/install/lib/python2.7/dist-packages/director/consoleapp.py", line 8, in <module>
    from director import applogic
  File "/home/liang/dev/drake-distro-1/build/install/lib/python2.7/dist-packages/director/applogic.py", line 5, in <module>
    import director.vtkAll as vtk
  File "/home/liang/dev/drake-distro-1/build/install/lib/python2.7/dist-packages/director/vtkAll.py", line 1, in <module>
    from vtk import *
  File "/home/liang/dev/drake-distro-1/build/externals/director/src/vtk-build/Wrapping/Python/vtk/__init__.py", line 41, in <module>
    from vtkCommonPython import *
ImportError: No module named vtkCommonPython
```
Searching for `vtkCommonPython`, I found a `.so` file but no directory of that name:

```
$ cd drake-distro
$ find . -name "vtkCommonPython*"
./build/externals/director/src/vtk-build/Common/vtkCommonPythonInit.cxx
./build/externals/director/src/vtk-build/Common/vtkCommonPythonInit.data
./build/externals/director/src/vtk-build/Common/CMakeFiles/vtkCommonPythonD.dir
./build/externals/director/src/vtk-build/Common/CMakeFiles/vtkCommonPythonD.dir/vtkCommonPythonInit.cxx.o
./build/externals/director/src/vtk-build/Common/CMakeFiles/vtkCommonPython.dir
./build/externals/director/src/vtk-build/Common/CMakeFiles/vtkCommonPython.dir/vtkCommonPythonInit.cxx.o
./build/externals/director/src/vtk-build/bin/vtkCommonPython.so
```

patmarion, 2016-10-07:
Ok, so to run it you should do these paths:

    LD_LIBRARY_PATH=/home/liang/dev/drake-distro-1/build/install/lib/vtk-5.10:$LD_LIBRARY_PATH
    PYTHONPATH=/home/liang/dev/drake-distro-1/build/externals/director/src/vtk-build/Wrapping/Python:/home/liang/dev/drake-distro-1/build/externals/director/src/vtk-build/bin:$PYTHONPATH

So we're setting `LD_LIBRARY_PATH` so that the dynamic loader can find the vtk libraries, it seems there's an RPATH issue there.  And we're adding two paths to the `PYTHONPATH`, one is to get the vtk python module (represented by `vtk/__init__.py`), and the other path is to get some additional vtk python shared objects (represented by `vtk*Python.so` files).  The system install on ubuntu organizes things so these paths aren't necessary, but it seems to be required for vtk's local install.  You can either export those environment variables or just put those settings on the command line preceding the directorPython executable.

patmarion, 2016-10-07:
By the way, it looks like the main issue regarding the python modules is that Director's superbuild installs vtk, but an extra install command is required to install vtk's python modules.  In my previous comment, the paths I gave you to include in `PYTHONPATH` point inside the vtk build dir, and not the vtk install dir.  But that should be ok for now.

liangfok, 2016-10-07:
:tada: It worked! Thanks!! I'm now able to run Drake Visualizer with ROS Kinect on Ubuntu 16.04. Before closing this issue, I'd like to codify all of the manual tweaks described above.

liangfok, 2016-10-09:
# Summary of How to Build Drake on Ubuntu 16.04 with ROS Kinect

[Drake](http://drake.mit.edu) currently uses the following version of [Director](https://github.com/RobotLocomotion/director), which is from [April 21, 2016](https://github.com/RobotLocomotion/director/pull/256):

* https://github.com/RobotLocomotion/director/tree/cfc7702bf92

Since Ubuntu 16.04 uses VTK 6 and Director requires VTK 5, we need to force Director to download and compile VTK 5 as an external. This is done using `CMake` flag `-DUSE_SYSTEM_VTK=OFF`. When this flag is applied, Director uses the [following version of VTK](https://github.com/RobotLocomotion/director/blob/cfc7702bf9278707bca82d0393bba8611513ea15/distro/superbuild/cmake/externals.cmake#L195), which was last updated March 4, 2015:

* https://github.com/bilke/VTK/tree/b753b7f

Note that the latest version of Director still uses the same version of VTK 5 when it does not use the system's VTK:

* https://github.com/RobotLocomotion/director/blob/master/distro/superbuild/cmake/externals.cmake#L196

Unfortunately, this version of VTK cannot be compiled because of the following error:

```
In file included from /usr/include/GL/glx.h:333:0,
                 from /home/liang/dev/drake-distro-1/build/externals/director/src/vtk/Rendering/vtkXOpenGLRenderWindow.cxx:31:
/usr/include/GL/glxext.h:480:143: error: ‘GLintptr’ has not been declared
 typedef void ( *PFNGLXCOPYBUFFERSUBDATANVPROC) (Display *dpy, GLXContext readCtx, GLXContext writeCtx, GLenum readTarget, GLenum writeTarget, GLintptr readOffset, GLintptr writeOffset, GLsizeiptr size);
                                                   
```

Fixing this error requires the following line to be uncommented as described [here](https://github.com/UV-CDAT/uvcdat/issues/983#issuecomment-71814770):

* https://github.com/bilke/VTK/blob/b753b7f67c9e9b161b2e783cfcac39af069fac2a/Rendering/vtkXOpenGLRenderWindow.cxx#L30

# Questions

1. Should Drake be updated to use the latest version of Director? Is the answer to this as simple as bumping the version used by Drake and then ensuring all of the unit tests and demos continue to run? Are there any reasons blocking us from updating the version of Director used by Drake?

2. Should I fork the version of VTK used by Director (https://github.com/bilke/VTK/tree/b753b7f), apply the [fix](https://github.com/UV-CDAT/uvcdat/issues/983#issuecomment-71814770) described above, then update Director to use my patched fork? Does Director have a set of unit and regression tests and CI slaves that I can run / use to verify that Director still works even after changing its version of VTK?

@patmarion, I'd appreciate your thoughts about these!

RussTedrake, 2016-10-09:
Also relevant #3647:  After looking around I was thinking that I'd try using VTK as the graphing/plotting tool even inside Drake since we already have the vtk dependency.  Obviously, I would like to use the most up to date vtk possible.

patmarion, 2016-10-09:
To answer 1., yes Drake should be updated to track director/master.  Normally, I manually bump drake's sha1 to director/master whenever needed.  Right now the divergence between the April 2016 version and master is pretty small.  I've tested recently and it's good to go now.

Yes, I agree with what you have described for 2, I think that would be a good move to resolve this quickly.  All pull requests to director will be tested on Travis CI, but they won't compile vtk from source, so your mods won't be tested.  It's a minor mod though, so I'm sure it's fine to go ahead with.

You said "this version of vtk cannot be compiled" but actually that version compiles fine, the compile error was when director includes it.  I'm glad that the fix works, but it makes me think there could be a fix that can be applied to director (since vtk compiles fine).

patmarion, 2016-10-09:
Also you said "Ubuntu 16.04 uses VTK 6", but I think that vtk5 is still available to install.  I thought the problem was that the newer version of ROS required vtk6, which conflicted with the vtk5 system install?

liangfok, 2016-10-09:
> Also you said "Ubuntu 16.04 uses VTK 6", but I think that vtk5 is still available to install. I thought the problem was that the newer version of ROS required vtk6, which conflicted with the vtk5 system install?

Good point. I updated my original post above to reflect this fact. Thanks!

liangfok, 2016-10-09:
@patmarion says:

> You said "this version of vtk cannot be compiled" but actually that version compiles fine, the compile error was when director includes it. I'm glad that the fix works, but it makes me think there could be a fix that can be applied to director (since vtk compiles fine).

Yes you're right. I updated my post above to reflect this idea.

PR #3738 (2016-10-10): Bumped director submodule to be the latest version.
This is necessary for Director to compile VTK5 from source on systems with ROS Kinetic installed. Note that this version of Director optionally uses a patched version of VTK5 that defines `GLX_GLXEXT_LEGACY`: https://github.com/RobotLocomotion/director/commit/38499093ba928a869bda48fb69437fa0f1a44189

Resolves #3703.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3738)
closed 2016-10-10 by liangfok