RobotLocomotion/drake #3710: kuka_demo fails during open source builds on OS X

jamiesnape, 2016-10-06:
```
Test command: /Users/elcapitan/workspace/mac-clang-ninja-nightly-matlab-open-source/build/drake/bin/kuka_demo
Welding joint iiwa_joint_ee
Welding joint tool0_joint
Welding joint iiwa_joint_ee
Welding joint tool0_joint
x = 0 0 0 0 0 0 0
x = 0 0 0 0 0 0 0
x = 0 0 0 0 0 0 0
x =    -0.384641     0.700006     0.400798     -1.92449     -1.40822     -1.07185 -7.25537e-12
x = 0 0 0 0 0 0 0
[2016-10-06 14:20:23.406] [console] [info] INFO[0] = 1 
[2016-10-06 14:20:23.407] [console] [info] INFO[1] = 100 
[2016-10-06 14:20:23.407] [console] [info] INFO[2] = 1 
[2016-10-06 14:20:23.407] [console] [info] INFO[3] = 1 
[2016-10-06 14:20:23.407] [console] [info] INFO[4] = 100 
libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: inverseKinPointwise failed to compute a valid solution.

Test #kuka_demo .......................................................................***Exception: Other  3.32 sec
```

david-german-tri, 2016-10-06:
Er...have we just been failing to buildcop the open-source build up to this point?  Looks like it's been yellow since September 14.

jamiesnape, 2016-10-06:
No idea. I just noticed as I was looking for collateral damage from the MOSEK outage.

jwnimmer-tri, 2016-10-06:
Oops.  I noticed this a day or two ago, and assumed it was some known regression that was being ignored by buildcop.  Sorry for not speaking up!

PR #3719 (2016-10-07): Require SNOPT or IPOPT to run kuka_demo
Fixes #3710 by disabling the test.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3719)
closed 2016-10-07 by jwnimmer-tri