RobotLocomotion/drake #3711: examples/Atlas/runAtlasDynamics fails during open source builds on OS X

jamiesnape, 2016-10-06:
```
 [Warning: This timestep violated our assumptions about which contacts could
possibly become active in one timestep.  Consider reducing your dt.  If it
seems to happen a lot, then please let us know about it.]
[> In WarningManager/warnOnce (line 21)
  In TimeSteppingRigidBodyManipulator/solveLCP (line 797)
  In TimeSteppingRigidBodyManipulator/update (line 204)
  In SimulinkModelHandle/sim (line 112)
  In DynamicalSystem/simulate (line 120)
  In runAtlasDynamics (line 27)]

status =

     0

Path 4.7.03: Matlab Link
2982 row/cols, 6219450 non-zeros, 69.94% dense.

Could not open options file: path.opt
Using defaults.
Path 4.7.03 (Fri Feb 10 15:00:24 2012)
Written by Todd Munson, Steven Dirkse, and Michael Ferris

INITIAL POINT STATISTICS
Maximum of X. . . . . . . . . .  0.0000e+00 var: (x[    1])
Maximum of F. . . . . . . . . .  1.2915e+08 eqn: (f[ 1428])
Maximum of Grad F . . . . . . .  2.3132e+02 eqn: (f[ 1121])
                                            var: (x[   25])

INITIAL JACOBIAN NORM STATISTICS
Maximum Row Norm. . . . . . . .  3.6992e+03 eqn: (f[ 1430])
Minimum Row Norm. . . . . . . .  1.4288e-02 eqn: (f[   65])
Maximum Column Norm . . . . . .  1.4193e+04 var: (x[   14])
Minimum Column Norm . . . . . .  4.0000e+00 var: (x[ 2494])

Crash Log
major  func  diff  size  residual    step       prox   (label)
    0     0             1.4465e+09             0.0e+00 (f[ 2406])
    1     2  1799  1106 1.4188e+09  8.0e-01    1.0e+01 (f[  553])
    2     3  1014  1678 1.3822e+09  1.0e+00    9.0e+00 (f[ 2407])
    3     4   450  1775 1.1795e+09  1.0e+00    8.1e+00 (f[  553])
    4     5   395  1766 1.0577e+09  1.0e+00    7.3e+00 (f[ 2407])
    5     8   301  1772 6.6590e+08  6.4e-01    6.6e+00 (f[  553])
    6    12   198  1710 3.8328e+08  4.1e-01    5.9e+00 (f[ 2407])
    7    17   142  1688 3.5546e+08  2.6e-01    5.3e+00 (f[  553])
    8    23    84  1658 2.3120e+08  1.7e-01    4.8e+00 (f[ 2407])
    9    30    32  1634 1.9767e+08  6.9e-02    4.3e+00 (f[ 2527])
   10    34    26  1606 1.9355e+08  4.1e-01    3.9e+00 (f[ 2527])
   11    38    31  1628 1.8603e+08  4.1e-01    3.5e+00 (f[ 2527])
   12    44    38  1644 1.8371e+08  1.7e-01    3.1e+00 (f[ 2527])
   13    50    33  1638 1.8113e+08  1.7e-01    2.8e+00 (f[ 2527])
   14    57    26  1653 1.7458e+08  6.9e-02    2.5e+00 (f[ 2527])
   15    65     6  1640 1.7369e+08  2.8e-02    2.3e+00 (f[ 2527])
pn_search terminated: no basis change.

Major Iteration Log
major minor  func  grad  residual    step  type prox    inorm  (label)
    0     0    66    16 1.7369e+08           I 2.1e+00 2.8e+07 (f[ 2527])
    1   490    67    17 8.7562e+07  1.0e+00 SM 8.2e-01 1.7e+07 (f[ 2527])
    2   200    68    18 3.7098e+07  1.0e+00 SO 3.3e-01 3.9e+06 (f[ 2920])
    3   179    69    19 1.4840e+07  1.0e+00 SO 1.3e-01 1.6e+06 (f[ 2683])
    4   190    70    20 5.0450e+06  1.0e+00 SO 5.3e-02 6.5e+05 (f[ 2703])
    5   242    71    21 1.6155e+06  1.0e+00 SO 2.1e-02 3.3e+05 (f[ 2705])
    6   260    72    22 3.7829e+05  1.0e+00 SO 8.4e-03 1.0e+05 (f[   18])
    7   148    73    23 1.9607e+05  1.0e+00 SO 3.4e-03 6.3e+04 (f[   18])
    8    96    74    24 1.1531e+05  1.0e+00 SO 1.3e-03 3.6e+04 (f[   18])
    9   136    75    25 8.9460e+04  1.0e+00 SO 5.4e-04 2.2e+04 (f[   18])
   10   133    76    26 3.8883e+04  1.0e+00 SO 2.2e-04 1.2e+04 (f[   18])
   11   288    77    27 2.3286e+04  1.0e+00 SO 8.6e-05 8.1e+03 (f[  558])
   12   388    78    28 1.0747e+04  1.0e+00 SO 3.5e-05 3.9e+03 (f[   71])
[... 321 lines cut ...]
            z = pathlcp(M,w,lb,ub);

Error in TimeSteppingRigidBodyManipulator/update (line 204)
        [obj,z,Mvn,wvn] = solveLCP(obj,t,x,u);

Error in SimulinkModelHandle/sim (line 112)
      varargout{:} = sim(obj.name,varargin{:});

Error in DynamicalSystem/simulate (line 120)
simout = sim(mdl,pstruct);

Error in runAtlasDynamics (line 27)
traj = simulate(sys,[0 2],x0);
Error using SimulinkModelHandle/sim (line 112)
Error reported by S-function 'DCSFunction' in 'MIMOCascade_3748998/system1/DrakeSys':

DrakeSystem S-Function: error PathLCP:FailedToSolve in MATLAB callback.
See additional debugging information above

Error in DynamicalSystem/simulate (line 120)
simout = sim(mdl,pstruct);

Error in runAtlasDynamics (line 27)
traj = simulate(sys,[0 2],x0);
```

RussTedrake, 2016-10-06:
Yikes.  The warning is clear enough; no idea why it would be related to an open-source build vs something else.

liangfok, 2016-10-12:
This issue continues to result in nightly CI failures:

* https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-clang-ninja-nightly-matlab-open-source/
* https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-clang-nightly-matlab-open-source/

jamiesnape, 2016-10-12:
I will look into it.

RussTedrake, 2017-02-03:
won’t fix.  aim is to replace this with a c++ version.

closed 2017-02-03 by RussTedrake