RobotLocomotion/drake #3713: Mysterious CI Failure - Undefined Reference to llvm_gcda_*

liangfok, 2016-10-07:
# The Error

```
../libplt.a(Xwin.c.o): In function `__llvm_gcov_writeout':
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x30fb): undefined reference to `llvm_gcda_start_file'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3123): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3137): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3158): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x316c): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x318d): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x31a1): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x31c2): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x31d6): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x31f7): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x320b): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x322c): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3240): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3261): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3275): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3296): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x32aa): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x32cb): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x32df): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3300): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3314): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3335): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3349): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x336a): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x337e): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x339f): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x33b3): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x33d4): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x33e8): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3409): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x341d): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x343e): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3452): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3473): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3487): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x34a8): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x34bc): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x34dd): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x34f1): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3512): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3526): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3547): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x355b): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x357c): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3590): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x35b1): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x35c5): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x35e6): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x35fa): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x361b): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x362f): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3650): undefined reference to `llvm_gcda_emit_function'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3664): undefined reference to `llvm_gcda_emit_arcs'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3669): undefined reference to `llvm_gcda_summary_info'
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x366e): undefined reference to `llvm_gcda_end_file'
../libplt.a(Xwin.c.o): In function `__llvm_gcov_init':
/home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3f76): undefined reference to `llvm_gcov_init'
collect2: error: ld returned 1 exit status
```

# Example Build

https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-clang-nightly-coverage/251/

# Prognosis

Could [avl](https://github.com/RobotLocomotion/avl/tree/06bffc46a4df4094b6a3c98f3dd191a54795fb96) be linking against the wrong shared libraries?

jamiesnape, 2016-10-07:
AVL does not link to any shared libraries other than X11 normally.

david-german-tri, 2017-01-03:
I'm going to presume this fixed in https://github.com/RobotLocomotion/drake-ci/pull/47.

closed 2017-01-03 by david-german-tri