RobotLocomotion/drake #3717: Matlab complains Snopt not found when running runAtlasWalking.m

chang-liang, 2016-10-07:
When I ran  runAtlasWalking.m in the Drake examples, Matlab complains like this:

`
[ 37%] Performing download step for 'snopt'
Cloning into 'externals/snopt'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights and the repository exists.
Clone of 'git@github.com:RobotLocomotion/snopt.git' into submodule path 'externals/snopt' failed
make[2]: *** [snopt-prefix/src/snopt-stamp/snopt-download] Error 1
make[1]: *** [CMakeFiles/snopt.dir/all] Error 2
make: *** [all] Error 2
`
Should I download a Snopt by myself? Thanks!

PS: 
I use:
Ubuntu 14.04 LTS ;
Matlab R2016a;
gcc 4.9;
g++ 4.9;

samkhal, 2016-10-09:
The lab cannot distribute the full version of snopt. You can use the free restricted version of snopt using my instructions: https://github.com/RobotLocomotion/drake/issues/3369#issuecomment-245497973.

chang-liang, 2016-10-12:
Problem solved! Thank you! @samkhal

closed 2016-10-12 by RussTedrake